webcamoid/webcamoid #476: Saveing color setting for Zoom possible?

stefan-franz, 2021-12-27:
Linux Mint 20 / Webcamoid 8 

Hello, 
is it possible, to save the colour setting for in the webcam i adjust with Webcamoid directly in driver for another applications like Zoom?
With Webcamoid i can fine adjust e.g. the colour and it would very useful, if i can save this anyway for other applications. 
Thanks in advance
Stefan

hipersayanX, 2021-12-27:
Do you mean this options?

[image]

yes, but those options are for the virtual camera, not per program. If you mean saving the settings like brightness, contrast, and so on, those options are per program, doesn't depends on the driver but it depends on the program using the virtual camera.
What do you want, saving settings per program, isn't technically possible.

stefan-franz, 2021-12-27:
I can't use the virtual camera option - i'm a linux newbie - what do i have to do?

hipersayanX, 2021-12-28:
@stefan-franz give me some context, what's the error message?

stefan-franz, 2021-12-28:
I use Linux Mint 20 - and Webcamoid 8.6.1

If i want to make a Virtual Camera, i have no entry - i'm not able to make it. 

[image]

hipersayanX, 2021-12-28:
Did you installed akvcam or v4l2loopback?

stefan-franz, 2021-12-28:
No, only the Webcamoid App. 
What of these 2 should i install? Do you have for me the link to the correct version for Linux Mint?

hipersayanX, 2021-12-28:
You can install [v4l2loopback](https://www.howtoinstall.me/ubuntu/18-04/v4l2loopback-dkms/) from your repositories, you can install akvcam using the [installer](https://github.com/webcamoid/akvcam/releases), both can be installed at same time and you can choose which one use, v4l2loopback is faster with almost no delay but with less options and less compatibility with some programs, akvcam is the official one but it could be a bit slower but with better compatibility and more options.

stefan-franz, 2022-01-03:
Today, i find a solution to adjust the color of the Webcam (white balance and hue is for me important) in Zoom:
Start Wecamoid and adjust in the source config the parameters. Shut down Webcamoid and switch in Zoom on the webcam. The system remembe the setting and in Zoom the color looks fine. 
But: It depends on the Webcam if it works - or not.

hipersayanX, 2022-01-04:
Oh, I understand, what you wanted to do is changing the settings of the webcam (the real one), for later usage in other programs, I thought you were talking about the virtual camera.
Anyways, when you turn off your computer or disconnect the camera, its settings will always go back to its defaults.

closed 2022-02-15 by hipersayanX