webcamoid/webcamoid #477: No camera preview shown

slhck, 2021-12-29:
# Reporting a problem?

## Summary

Write here a brief description of the problem.

## Current Behavior

When opened, Webcamoid does not show any camera preview:

[image]

The same goes for a third-party camera connected via USB-C:

[image]

The cameras themselves work (as tested with FaceTime, OBS, …) and their lights are on.

## Expected Behavior

I am expecting to see a camera preview.

## Steps to Reproduce

1.  Open Webcamoid
2.  Select any camera

## Your Environment

* Webcamoid information: Daily build from 29.12.2021
* Operating System information: macOS 12.1 (21C52), ARM64
* MacBook Pro M1 with built-in FaceTime camera

hipersayanX, 2021-12-29:
Let say you select the YUY2, and you switch **Play sources** on and off, it still does not works?

slhck, 2021-12-29:
I switched to the Logitech cam and selected YUV2. Switching that toggle does nothing; it also doesn't really look like it's on or off:

[image]

hipersayanX, 2022-01-06:
Can you test this and #478 in the daily build? hopefully it should be working now.

slhck, 2022-01-06:
I can now see the preview but it's all blue:

[image]

I'm putting up grayscale cards so you get the idea.

This is happening with all cameras, both for the FaceTime camera with the v024 color format as well as the Logitech StreamCam with YUV2.

hipersayanX, 2022-01-08:
Now I've switched the data capture format to 24RGB, maybe it should work now, maybe. In my case its already working fine, so not sure at all what's causing it on your side.

slhck, 2022-01-09:
Yes, it works now, thank you!

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX