webcamoid/webcamoid #478: App hangs when changing cameras

slhck, 2021-12-29:
# Reporting a problem?

## Summary

When I switch from one camera to another, the app hangs.

## Current Behavior

I open the application and switch from one camera to another. The beach ball appears and the app hangs. I have to force-quit it.

## Expected Behavior

Simply switch to another camera.

## Steps to Reproduce

1.  Open the app
2.  Click the Settings --> Video button
3.  Click the camera name to enter its settings
4.  Press X to close the settings
5.  Choose another camera
6. App hangs

## Your Environment

* Webcamoid information: Daily build from 29.12.2021
* Operating System information: macOS 12.1 (21C52), ARM64
* MacBook Pro M1 with built-in FaceTime camera

## Crash log

<details>
<summary>Crash log</summary>
<pre>
Date/Time:        2021-12-29 19:53:52.788 +0100
End time:         2021-12-29 19:55:57.215 +0100
OS Version:       macOS 12.1 (Build 21C52)
Architecture:     arm64e
Report Version:   35
Incident Identifier: F72959AE-3123-4AF6-A930-0035A2DB9647
Share With Devs:  Yes

Data Source:      Stackshots
Shared Cache:     66C33C62-48A9-31E5-B8B5-7A71C9760D07 slid base address 0x1acd00000, slide 0x2cd00000
Shared Cache:     E72A2011-6ACF-3F25-BFE7-730570330401 slid base address 0x7ff81f464000, slide 0x1f464000
Shared Cache:     7B87B74E-726C-3672-8699-6F7BEAA1E114 slid base address 0x193b28000, slide 0x13b28000

Command:          Webcamoid
Path:             /Applications/Webcamoid.app/Contents/MacOS/Webcamoid
Identifier:       com.webcamoidprj.webcamoid
Version:          9.0.0 (9.0.0)
Architecture:     x86_64 (translated)
Parent:           launchd [1]
PID:              29671
Time Since Fork:  239s

Event:            hang
Duration:         124.43s
Duration Sampled: 1.20s (process was unresponsive for 123 seconds before sampling)
Steps:            12 (100ms sampling interval)

Hardware model:   MacBookPro18,3
Active cpus:      8
HW page size:     16384
VM page size:     16384

Time Awake Since Boot: 34640s
Time Since Wake:  3535s

Fan speed:        0 rpm
[... 269 lines cut ...]
           *6   psynch_cvcontinue + 0 (pthread + 18048) [0xfffffe000a532180] 7-12

  Binary Images:
           0x100261000 -        0x100418fff  com.webcamoidprj.webcamoid 9.0.0 (9.0.0)       <C95CA3DB-6CC4-331E-A1C3-EE0B0ED6A4E4>              /Applications/Webcamoid.app/Contents/MacOS/Webcamoid
           0x109622000 -        0x109ab5fff  org.qt-project.QtWidgets 5.15 (5.15.2)         <D801C509-1422-3755-9DF9-31E58A9EF464>              /Applications/Webcamoid.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets
           0x10a3cc000 -        0x10a75ffff  org.qt-project.QtQuick 5.15 (5.15.2)           <1FC555D0-DFF9-3185-A081-B09548D78ACE>              /Applications/Webcamoid.app/Contents/Frameworks/QtQuick.framework/Versions/5/QtQuick
           0x10ae7f000 -        0x10b1f6fff  org.qt-project.QtQml 5.15 (5.15.2)             <4D3930B8-1D10-3B8E-B0EC-FA5B0A3652BF>              /Applications/Webcamoid.app/Contents/Frameworks/QtQml.framework/Versions/5/QtQml
           0x10bbc6000 -        0x10bca5fff  org.qt-project.QtQuickTemplates2 5.15 (5.15.2) <6975C29E-6ADD-3EB8-9F4F-A40D662CED88>              /Applications/Webcamoid.app/Contents/Frameworks/QtQuickTemplates2.framework/Versions/5/QtQuickTemplates2
           0x10c1b9000 -        0x10c6d0fff  org.qt-project.QtGui 5.15 (5.15.2)             <DA53DB54-97B3-33E4-B043-8AB4DB184908>              /Applications/Webcamoid.app/Contents/Frameworks/QtGui.framework/Versions/5/QtGui
           0x10d063000 -        0x10d1e2fff  libqcocoa.dylib (0)                            <E1670066-A1CE-31B1-9B62-6B81ED18C519>              /Applications/Webcamoid.app/Contents/Plugins/platforms/libqcocoa.dylib
           0x10d66d000 -        0x10dbe4fff  org.qt-project.QtCore 5.15 (5.15.2)            <7046633A-7ED1-3A02-B6F9-69F438EDC363>              /Applications/Webcamoid.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore
           0x1125e0000 -        0x1125fbfff  libVideoCapture_avfoundation.dylib (0)         <4B001295-8BD4-3D08-8D95-490C0D151C85>              /Applications/Webcamoid.app/Contents/Plugins/avkys/libVideoCapture_avfoundation.dylib
           0x11354c000 -        0x113567fff  libVideoCapture.dylib (0)                      <0701BB13-30B2-3EDB-8D7C-59D2B608A0F6>              /Applications/Webcamoid.app/Contents/Plugins/avkys/libVideoCapture.dylib
           0x20045a000 -        0x2004ddfff  dyld (941.4)                                   <CEF5A27A-D50B-3020-AF03-1734B19BC8C5>              /usr/lib/dyld
        0x7ff7fff69000 -     0x7ff7fffa8fff  runtime (254.5.4)                              <21C1E0C9-A36E-3E4B-A12B-1BF54CE4403E>              /usr/libexec/rosetta/runtime
        0x7ff81f77b000 -     0x7ff81f786fff  libsystem_pthread.dylib (485.60.2)             <6C7561B4-4B92-3F45-921E-ABE669299844>              /usr/lib/system/libsystem_pthread.dylib
        0x7ff81f7c8000 -     0x7ff81fcc8fff  com.apple.CoreFoundation 6.9 (1856.105)        <7E1D1901-3F9E-3E2E-A090-3655E5F5E04B>              /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
        0x7ff82215d000 -     0x7ff822febfff  com.apple.AppKit 6.9 (2113.20.111)             <D23B9681-3764-3298-A716-FBB511DD5A7C>              /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
        0x7ff82740c000 -     0x7ff82745dfff  com.apple.CoreVideo 1.8 (0.0)                  <127E441F-3B6D-3C27-8A68-501B156FF6B9>              /System/Library/Frameworks/CoreVideo.framework/Versions/A/CoreVideo
        0x7ff828638000 -     0x7ff82865afff  com.apple.audio.caulk 1.0 (91)                 <E76D1AB3-89C2-3A93-990B-30827FCF89B0>              /System/Library/PrivateFrameworks/caulk.framework/Versions/A/caulk
        0x7ff828895000 -     0x7ff828b8cfff  com.apple.HIToolbox 2.1.1 (1113.6)             <535DBC28-4D26-3609-86BB-5C40227B648C>              /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox
   *0xfffffe0007918000 - 0xfffffe000818bfff  kernel.release.t6000 (8019.61.5)               <2F4E22DC-64BE-3369-B574-0909F0571FA6>__TEXT_EXEC   /System/Library/Kernels/kernel.release.t6000
   *0xfffffe000a52db00 - 0xfffffe000a533bc7  com.apple.kec.pthread 1.0 (1)                  <C89316C8-32EB-3CFE-B1E0-48F24DD8F3FF>__TEXT_EXEC   /System/Library/Extensions/pthread.kext/Contents/MacOS/pthread
</pre>
</details>

slhck, 2022-01-06:
This is now fixed in the daily build.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX