webcamoid/webcamoid #479: Virtual camera error : can't create a vcam

Hydr0Btz, 2022-01-01:
Hello guys,
I'm here because i have a problem since i installed webcamoid : i want to create a vcam to have effects on websites such as omegle or for obs studio. I'm using kubuntu lts 20.04 and my webcamoid version is 8.8.0. i already installed v4l2loopback. the problem i have might be related to #273 

The problem is that when i want to create a virtual cam it doesn't show me anything in the first chosing space like that (even when i click)
[image]
And then i try to click on add and then i chose a name and enter my password and it tells me an error with no description :
[image]
 I really want those issue to be resolved asap so if you know how to setup a vcam that would help me a lot !

hipersayanX, 2022-01-04:
You can try with the daily build, I've improved the error messages, it can probably give a better clue of what is happening.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX