webcamoid/webcamoid #480: Crash on launch since Linux Mint update

claudemontreal, 2022-01-08:
The program was working fine before. Since I've updated Linux Mint to Una, the program crashes on startup. I've uninstalled it, went https://webcamoid.github.io/ and tried all the three different versions - installable, appimage and portable. Same result with the three: the program launches for two seconds, then it crashes.

In terminal, it says this:
qrc:/Webcamoid/share/qml/RecordConfig.qml:126:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/VirtualCamera/share/qml/main.qml:25:1: QML GridLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
Segmentation fault (core dumped)

hipersayanX, 2022-01-08:
Did you tried the daily build? just in case, are you using Wayland?

claudemontreal, 2022-01-08:
> Did you tried the daily build? just in case, are you using Wayland?

I have no idea what those are unfortunately. I'm just a regular user wanting the program that I was using to work, I'm not a developer or anything.

hjns, 2022-01-13:
I'm running Mint Una (updated daily as the update manager finds updates) with the Cinnamon desktop manager and haven't had any crashes. I'm running the AppImage version of 8.8.0.

mttr402, 2022-01-13:
Tried on Mint 19.1 and non of them work I get this error
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/main/webcamoid/lib/x86_64-linux-gnu/libQt5Quick.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/main/webcamoid/lib/x86_64-linux-gnu/libQt5Widgets.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/main/webcamoid/lib/x86_64-linux-gnu/libQt5Gui.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/main/webcamoid/lib/x86_64-linux-gnu/libQt5Qml.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/main/webcamoid/lib/x86_64-linux-gnu/libQt5Core.so.5)
webcamoid: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /home/main/webcamoid/lib/x86_64-linux-gnu/libQt5Core.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/main/webcamoid/lib/x86_64-linux-gnu/libpng16.so.16)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/main/webcamoid/lib/x86_64-linux-gnu/libicui18n.so.66)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/main/webcamoid/lib/x86_64-linux-gnu/libicuuc.so.66)

hipersayanX, 2022-01-13:
@claudemontreal @hjns check the [daily build](https://github.com/webcamoid/webcamoid/releases/tag/daily-build), the bug you are reporting here could already be resolved there.
@mttr402 you need Ubuntu Focal Fossa (Linux Mint Ulyana 20) at least.

gschwaer, 2022-01-13:
Hi @claudemontreal, have you tried to [reset the config](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh)? That worked for me.

claudemontreal, 2022-01-14:
> @claudemontreal @hjns check the [daily build](https://github.com/webcamoid/webcamoid/releases/tag/daily-build), the bug you are reporting here could already be resolved there. @mttr402 you need Ubuntu Focal Fossa (Linux Mint Ulyana 20) at least.

Hi, I just installed the daily build. It doesn't crash but it's not the same program than on https://webcamoid.github.io/. This last one is the program I want to use.

claudemontreal, 2022-01-14:
> Hi @claudemontreal, have you tried to [reset the config](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh)? That worked for me.

This worked. Now it isn't crashing on launch anymore. Thank you.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX