webcamoid/webcamoid #481: Can I use a USB camera with Webcamcoid?

guercini, 2022-01-14:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debugger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates below that fit better your issue.

# Reporting a problem?

## Summary

Write here a brief description of the problem.

## Current Behavior

Describe the problem the best as you can, don't omit information.

## Expected Behavior

How it should have work?

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue below:

1.  Open this
2.  Click that
3.  Drag those
4.  ...

## Suggestions and tips

How would you solve the problem?

## Your Environment

* Webcamoid information: (version) (architecture)
* Operating System information: (name) (numeric version and codename if applied) (architecture)
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)

# Want a new feature?

Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questions and other matters?

Go ahead!
Can  use a USB camera with Webcamcoid? If so, what are the steps to setup the camera to capture images? So far the camera is connected to the computer (Windows 10 Pro) via USB cable and Webcamoid is installed.

hipersayanX, 2022-01-14:
>Can I use a USB camera with Webcamcoid?

Yes

>If so, what are the steps to setup the camera to capture images?

Just connect the camera and click the play button.

guercini, 2022-01-14:
Thanks for the quick reply. I don't see the play button. Here is what I have:

[image]

hipersayanX, 2022-01-14:
The Play/Stop button is the one in the left side, I've market it in the capture bellow:

[image]

Also, it seems to be detecting your webcams just fine, are not it working?

guercini, 2022-01-14:
Thanks again for the comments. I think I will install the Webcamoid on my other computer and try it again using your instructions. This computer has too much stuff on it.

hipersayanX, 2022-01-14:
Ok, report if you get it working, if the cameras are detected or not, if detected but they are working or not.

closed 2022-02-15 by hipersayanX