webcamoid/webcamoid #482: Switching video output format to WebM crashes program and stops it from opening again

cooljpeg, 2022-01-17:
I switched my video output format to WebM just to give it a whirl, but the program immediately crashed and now I cannot even open the program without it immediately crashing after 1 second of being open, basically leaving me totally locked out. Is this a known issue? Thank you so much in advance! :+}

Version: 8.8.0
OS: Windows 10 Home, Build 19043.1466

hipersayanX, 2022-01-18:
Try [resetting configs](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh), also you can try the [daily build](https://github.com/webcamoid/webcamoid/releases/tag/daily-build) and check if its still happening .

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX