webcamoid/webcamoid #483: Installation error in Linux Mint

OliverStunt, 2022-01-17:
## Summary

Webcamoid refuses to install. Critical error breaks the installation via Terminal.

## Current Behavior

I tried to excecute the files and run them as a program, via instructions from the web (absolute Linux newbie here).
I tried installing it with commands listed in the error reports. Couple of trie sof each brought always reproducible and exact same results.
Version 6.7.0 from mintinstall works fine, except from minor bugs purportedly fixed in new versions.

## Expected Behavior installation

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue below:

1) Selecting and downloading one of the appropriate... builds? files? from this page
2) Selecting property: allow to excecute from Properties window
3) Installer opens, goes through well up to a point, messages in Terminal appear, installation stops and nothing changes.

## Suggestions and tips ---

## Your Environment

* Webcamoid information: 8.8.0 x64
* Operating System information: Linux Mint 20.2 Cinnamon 5.0.7  64-bit
* Any other useful information: file with errors and feedback attached

# Want a new feature? ---
# Questions and other matters? ---

Thank You all so much for all Your hard work! Maybe one day I'll code good enough to help back!
But a lot time will 
[error.txt](https://github.com/webcamoid/webcamoid/files/7885126/error.txt)
have to pass first xD

hipersayanX, 2022-01-18:
It seems to be a bug in the installer, just try with the AppImage or the tar.xz package, also you can check if the installer is still failing in the [daily build](https://github.com/webcamoid/webcamoid/releases/tag/daily-build).

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX