webcamoid/webcamoid #487: Daily build description

HelenFoster, 2022-01-29:
Hi, I was looking at the daily build page
https://github.com/webcamoid/webcamoid/releases/tag/daily-build
but at first I thought it was broken because it says it was released 15 Jul 2021 with 277 commits to master since this release. It does mention yesterday's commit in the description, so I guess the assets are for that after all? I'd suggest adding some text that tells us to ignore the release date, if that's the case.

hipersayanX, 2022-01-29:
Yeap, I know, other users also asked for the same, the problem is that I can't keep re-tagging the daily build with every new commit, so the best solution I've found was creating a fixed tag and updating the files with each new commit, while pointing to the right commit in the description.

>I'd suggest adding some text that tells us to ignore the release date, if that's the case.

Will see what can I do with that, there are not much space to put text in [there](https://github.com/webcamoid/webcamoid/blob/581d1d6506a47b421872e199610a54f833a8f568/.github/workflows/linux-ubuntu.yml#L130) :neutral_face:

HelenFoster, 2022-01-29:
Even something really short would help, like `"Built from commit ${{ github.sha }} (not the tagged one). ${{ github.event.head_commit.message }}"`

hipersayanX, 2022-01-31:
Does it looks better now?

HelenFoster, 2022-01-31:
Thanks, that should do it!

closed 2022-01-31 by HelenFoster