webcamoid/webcamoid #488: "Add effect" can hang on default ASCII Art

HelenFoster, 2022-01-29:
## Summary

“Add effect” attempts to show the ASCII Art effect right away, and hangs if this is too much work.

## Current Behavior

When I click “add effect”, it starts by showing the ASCII Art effect. It lags trying to display this, and it’s a fight to change to a different effect. If I try this with increasing camera resolution, it becomes worse, until at some point it’s impossible to escape and I have to terminate the program.

At resolutions for which I can change to another effect, all the other effects I tried worked more smoothly.

## Expected Behavior

Ideally it would be able to bail out gracefully if a certain effect is too much to ask. If it can’t do this, it would be better not to show any effect until I click on it (so if there is a problem, I can avoid clicking that one next time).

## Steps to Reproduce

After first opening the program, while the camera is running, click the effects button and then “add effect”. It shows the ASCII Art effect right away. If the camera resolution is too high and the computer too slow, it hangs here.

## Your Environment

* Webcamoid information: f16ebc4 win64
* Operating System information: Windows 10.0.19042 64-bit

hipersayanX, 2022-01-31:
It happens when using the Natural mode, Fixed mode is working fine, the problem is that its drawing the image every time for each character over and over ([here](https://github.com/webcamoid/webcamoid/blob/f16ebc46fe0c05bdef77f84770186e7b13219527/libAvKys/Plugins/Charify/src/charifyelement.cpp#L265)), I was thinking that maybe a solution would be, pre-calculating the images for each character, then create a mask based on the down gray scaled version of  the source, were each pixel is converted to its equivalent character (using the previous table), then apply the mask to either the foreground color (Fixed) or the down scaled image (Natural), then draw the result in the background.
But I'll work on it after releasing Webcamoid 9.0.0, I want to do it quietly and without haste .

HelenFoster, 2022-01-31:
Thanks for your reply. I agree that fixing ASCII Art properly can wait. The issue here is that it's interfering with my ability to use other effects.

I don't actually want to use ASCII Art. Switching it into Fixed mode doesn't solve the problem, because as soon as I remove the effect, it forgets, and the problem returns.

For an immediate fix, could you make Fixed mode the default?

hipersayanX, 2022-01-31:
I've set Fixed mode as default for now, I'll keep the issue open.

HelenFoster, 2022-02-01:
I installed the latest daily build and deleted the registry key to make sure, and it still seems to be defaulting to Natural.

hipersayanX, 2022-02-02:
Are you sure you deleted the registry keys with Webcamoid closed? because I've tested before committing, and it was working fine for me, and is the exact same code for everyone else.
If you try to delete the key while Webcamoid is still running it won't work.

HelenFoster, 2022-02-03:
Looks like the installer I downloaded was the same as the one I downloaded before. Still need to check if I had a cache issue or something.

hipersayanX, 2022-02-03:
Check latest daily build. I was testing in Android and as you said, it was totally unusable, so I decided to work on it, it should work at an acceptable frame rate now, hopefully.

HelenFoster, 2022-02-07:
The Natural ASCII Art is now working for me at reasonable resolutions.

HelenFoster, 2022-02-07:
Though it's still more resource-intensive than many of the other effects, and I can still provoke it into hanging in cases where other effects would be OK.

So I still think it would be better if the software didn't attempt to preview it until it was clicked on.

hipersayanX, 2022-02-08:
Out of curiosity, with which resolution are you testing the effect, and what are you CPU specs?

>So I still think it would be better if the software didn't attempt to preview it until it was clicked on.

But then users will complain the effect preview is not working :confused: 

For the moment I will make a new release as is right now, at least with all other bugs fixed, and will dedicate time to this bug later.

HelenFoster, 2022-02-08:
> But then users will complain the effect preview is not working :confused:

I'm not sure if I explained well enough. Currently, this window opens with ASCII Art selected. I was thinking it could be initialized with nothing selected. Would that be confusing?

[image]

hipersayanX, 2022-02-08:
>Sorry, I explained that badly. Currently, this window opens with ASCII Art selected. I was thinking it could be initialized with nothing selected. Would that be confusing?

But even if there are no effects selected, once the user selects the ASCII Art effect, if meet conditions, it will trigger the hang, and the user **will undeniably** do so, and there we are at same point :sweat: 

>I'm currently stress-testing on a Core i3-6100 at 30fps.
>At 1920x1080, ASCII Art was fixed with your latest update.
>At 2592x1944, it can do Invert, but ASCII Art hangs.
>This camera goes up to 3264x2448 but I'm not going to try it.

More or less the same as me, I've a Core i3-3120M 2.50GHz (a 10 years old CPU), and tested it with a 4K video and it lags (a lot) but it does not hangs the program, tested in GNU/Linux.

>I was thinking, ideally for the general case, it would detect any situation where it can't keep up with what it's trying to do and stop trying (rather than hanging). I know that's a big job of course.

The diagram for the effects chain is more or less like this

```
Source -Ⓐ-> Effect1 -> Effect2 -> Effect3 -Ⓑ-> Screen
```

I was thinking maybe on measuring the time between the frames entering to Ⓐ and the frames coming out of Ⓑ, and if it takes 1 seconds or a bit more (< 1FPS), disconnect Ⓐ and advice the user to either continue or disable the effects, but it will take some time to design it and test it well, so I prefer to leave this issue open and fix it for 9.1.0.

HelenFoster, 2022-02-09:
If I go in the camera options and the "FPS" dropdown menu has a single option of "30", does that mean the camera outputs 30 FPS? If I then set an effect chain which can run at 10 FPS, what does it do? Is it any different if I'm recording? Does a hang on the preview screen mean that a single frame is taking longer than I'm willing to wait? (In some cases I could click another effect and it would change maybe 30 seconds later.)

hipersayanX, 2022-02-09:
The more or slower effects you add to the chain, the lower FPS you will get. On recording the video, the video will be marked with the same FPS as the camera (or whatever source you are using), but the video will be recorded at the real frame rate, for example, if you are recording your camera at 30 FPS (as market in the frame rate control), but the capture is running slower either because you are using some effects, or just because the camera is grabbing frames slower, let say at 15 FPS, you will get a video marked as 30 FPS, but when playing it you will get a 15 FPS video.
Putting everything on simple words, what you get on screen is what you will get on video recording.

HelenFoster, 2022-02-11:
I see! I'm looking at it again on 40214be (win64). At 2592x1944, ASCII Art now shows the first few frames reasonably quickly (and if I click another effect right away it can escape), but if I watch it for a few seconds the frame rate drops before heading into the hang. Sounds like it's thrashing actually. I have 8G of RAM on that machine.