webcamoid/webcamoid #491: Daily build not updating?

HelenFoster, 2022-02-03:
I'm downloading `webcamoid-installer-windows-daily-master-1-win64.exe` and getting an identical file as a few days ago. I tried a private window and a different web browser.

hipersayanX, 2022-02-03:
The actions are informing that the files are being updating :confused: 

https://github.com/webcamoid/webcamoid/runs/5057841560?check_suite_focus=true

You can check the **Daily Build Upload**.

hipersayanX, 2022-02-03:
Oh, my fault (kinda), it wast not creating the installer, that's why you were getting an old build. The portable package is working fine.

HelenFoster, 2022-02-04:
> May I add the commit hash in the About page to be sure?

I think that's a good idea. I looked for it there first, and didn't notice it was in the other place.

hipersayanX, 2022-02-05:
Ok, it's fixed, and also have added the commit hash in the title and the About page.

HelenFoster, 2022-02-07:
Great, thanks!