webcamoid/webcamoid #495: [Advice] Which backend is preferred?

lgbaldoni, 2022-02-16:
Packaging 9.0.0 for openSUSE Tumbleweed x86_64 and seeing both gstreamer and libav support: which should be used, ideally? Or should it be linked to both? In that case, which one will be preferred at runtime?

P.S: is the IRC channel still active somewhere?

hipersayanX, 2022-02-16:
>which should be used, ideally?

FFmpeg/libav for recording, VLC for playback.

>Or should it be linked to both?

Ideally it must be linked to both, and let user choose which one they want at run time.

>In that case, which one will be preferred at runtime?

FFmpeg/libav for recording, VLC for playback.

lgbaldoni, 2022-02-16:
Ok, but what about gstreamer?

hipersayanX, 2022-02-16:
You deleted your message, but I will answer anyway. Just compile with support for both, GStreamer and FFmpeg, they are not mutually exclusive.

lgbaldoni, 2022-02-17:
@hipersayanX ok, for now I'm leaving gstreamer out or there will be too many dependencies to install for a single package.

hipersayanX, 2022-02-17:
If RPM allow optional dependencies (with for sure, it allows) just put GStreamer as an optional dependency, no need to remove GStreamer at all.

lgbaldoni, 2022-02-17:
> If RPM allow optional dependencies (with for sure, it allows) just put GStreamer as an optional dependency, no need to remove GStreamer at all.

@hipersayanX libgstreamer will be pulled in if linked. Unsure about what else might be required.

closed 2023-05-11 by hipersayanX