webcamoid/webcamoid #497: 9.0.0 goes into "not responding" immediately when opening it

krillep, 2022-02-21:
## Current Behavior

After installing Webcamoid (9.0.0), every time it gets opened, it goes into "not responding" immediately. I've tried closing all background programs (Steam, Battle.net, Discord, etc). Last used version (don't remember number) worked fine back when I used it (couple months ago)

## Steps to Reproduce

1. Open
2. Cry
3. Force Close

## Your Environment

* Webcamoid information: version 9.0.0
* Operating System information: Windows 11 Pro 64-bit, build 22000

* Specs -
Motherboard: ASUS ROG Strix B550-F Gaming
CPU: AMD Ryzen 7 3700x
GPU: ASUS ROG Strix RX 5700 XT OC
Memory: G.Skill TridentZ NEO RGB DDR4 32GB 3600 Mhz
Storage: Sabrent Rocket 4.0 1TB M.2 NVMe

I could see there was a similar issue fixed some days ago, albeit with version 8.7.4 of the portable one.

krillep, 2022-02-21:
Culprit seems to be the virtual camera driver. As soon as it's installed, Webcamoid starts freezing upon starting it. If it's not installed, Webcamoid runs fine. When I try to install the driver via Webcamoid instead of manually, it also seems to fail every time.

hipersayanX, 2022-02-22:
>but there's no folder for the driver in its usual place (assuming Program Files

NSIS does not informs you where the driver were installed?

krillep, 2022-02-22:
Oddly enough, after the reinstall I just did, the program worked on the first start afterwards. After shutting it down and back up again, it went into constant "not responding" again :x

hipersayanX, 2022-02-22:
>and then asks if I want to install without asking me where - I press Yes, and then it seems to install for a few seconds - after that, nothing,

So it's not showing the NSIS installer dialog? (the one you see when installing manually)

>Edit: I noticed that there are two "AkVCamManager" background processes that open every time I try to launch the program.

Yes, that's ok, in the task manager you must see only one unique AkVCamAssistant process, and there can be more than one AkVCamManager process. The AkVCamManager is used to manage the vcams and to listen events from other clients, when opening Webamoid there is always a AkVCamManager process listening for vcam events.

>Edit 2: If I open more than one instance of Webcamoid, it seems only the first one hangs with "not responding". As long as I have that one open, every other start of Webcamoid runs smoothly it seems. This seems to be 100% consistent so far. If I close all instances, the first one I open again will go into "not responding" :x

Other users also informed me bugs when running many instances of Webcamoid. I will make it single instance only.

>Edit 4: Browser does not recognize the virtual driver (tried both Firefox and Edge)

I tested both just before release and it was working fine for me :confused: 

Also, thanks for the information :smiley:

krillep, 2022-02-23:
> So it's not showing the NSIS installer dialog? (the one you see when installing manually)
It does when installing manually, but not when installing through Webcamoid.

So I tried installing the Daily version (devel-1) instead of the regular one - at first there was not really any difference, but then I tried deleting the "x86" folder under the driver, and for a while it worked perfectly - outputs worked, browsers could see the virtual driver, it booted normally. Then I tried going back to regular version to see if it also worked there or was simply a fluke, but it still didn't work there.
Went back to Daily version, and now it's just as buggy lol. I have no idea what's going on :x

Also, I don't know if this has any impact - the regular version tries to install to Program Files, while the Daily version installs it to Program Files (x86)

matallui, 2022-02-25:
I'm having the same issue. Webcamoid works fine until I install the virtual camera. The installation via the app didn't work either, so I installed the virtual camera manually. Now Webcamoid just goes straight to "Not responding".

I've also noticed similar problems on MacOS.

FunFreshNew, 2022-03-03:
Same issue here(maybe). Previously I changed the filters to face tracking and suddenly the software is really laggy and ended up freezing. I think the problem for me is the file configuration executed the same config and makes the app didn't run properly every time. 
But I haven't found the folder/file for the configuration. So, where should I find the file that contains the config for the software? 
Also previously I did re-install and delete its registry, all folder and files that contain "webcamoid". But still, no changes.

hipersayanX, 2022-03-03:
@FunFreshNew 

>But I haven't found the folder/file for the configuration.

The configurations are in the registry under _HKEY_CURRENT_USER\Software\Webcamoid_.

dpkristensen, 2022-03-16:
Hi, just installed this for the first time and running into this as well.  A couple of things I found:

* Worked fine until I installed AkVirtualCamera
* AkVcamManager version is 9.1.0 (-p -v)
* No camera devices existed, so I created one in an admin shell with the steps on https://github.com/webcamoid/akvirtualcamera/wiki/Configure-the-cameras
* I see in process explorer that it's got 2 AkVCamManager child processes open.  One is "C:\Programs\AkVirtualCamera\x64\AkVCamManager.exe -p listen-events", and the other looks like it's going through a sequence of commands to get information and start the stream; I think the last commands were "stream ..." and "-p -v"

Two problems with this:
* Running the commands shouldn't block the main event thread, as this will cause windows to display "not responding".
* AkVCamManager takes a long time to start for some reason, on the order of seconds.

```
PS C:\Programs\AkVirtualCamera\x64> Measure-Command {start-process AkVCamManager -argumentlist "-p -v" -wait}

Days              : 0
Hours             : 0
Minutes           : 0
Seconds           : 11
Milliseconds      : 109
Ticks             : 111095588
TotalDays         : 0.000128582856481481
TotalHours        : 0.00308598855555556
TotalMinutes      : 0.185159313333333
TotalSeconds      : 11.1095588
TotalMilliseconds : 11109.5588
```

dpkristensen, 2022-03-16:
Looking around the manager code, I see a couple issues with https://github.com/webcamoid/akvirtualcamera/blob/master/dshow/PlatformUtils/src/messageserver.cpp:

* It's calling the 32-bit API explicitly, which might incur some virtualization overhead for the 64-bit native program.
* Every command, even -v incurs the overhead of initializing the message server.  Some information I found at https://github.com/tomerfiliba-org/rpyc/issues/306 indicates that named pipes can be VERY slow for some reason.  I'm not sure if this issue is related, but the times shown there do seem to be on the same order as what I observed above.

ihavenonickname, 2022-03-26:
It's eating memory like crazy. I took this screenshot like 2 minutes after opening the program for the first time.

[image]

When it reached 7.9 gb, a few seconds later, I gave up and killed the process.

I'm using a Windows 10 machine with a Ryzen 5 3600 cpu.

ihavenonickname, 2022-03-26:
After a few tests I noticed that some effects make the memory consumption skyrocket, like blur and fire. But it's not every effect. Black & white, for instance, looks okay.

ihavenonickname, 2022-03-31:
@hipersayanX 

> It would be useful to know which was the resolution of the video in which you tested the effect

1280 x 720

> it depends if the memory go increasing or its stable

It increases non-stop, and very fast. Every few seconds it eats a few hundred megabytes.

cjbarth, 2022-04-06:
I just tested this with the latest release of Webcamoid 9.0.0 for Windows, Portable 64-bit from https://webcamoid.github.io/ and found the exact same behavior. It works fine before installing AkVirtualCamera, and never works again after the driver is installed. FWIW, I'm also seeing the memory climb crazy fast. I'm seeing memory using for Webcamoid increase at the rate of about 10 MB/sec even though Windows reports it as "Not Responding"; it has already used 12.5 GB.

hipersayanX, 2022-04-06:
@cjbarth Try commenting [this line](https://github.com/webcamoid/webcamoid/blob/694c23e8ff9a6dbbb3d2168456ddc9c9d4489c77/libAvKys/Plugins/VirtualCamera/src/dshow/src/vcamdshow.cpp#L1179) and check if the memory still grow.

DTayloe, 2022-04-10:
Looks like to me the virtual camera component might be causing the issues. It failed to install the second time, because it could not overwrite the existing installer that was currently running the the background. When I ran the virtual webcam installer manually, it asked for admin privileges which I didn't see when installing from Webcamoid. Maybe simple permissions issue? I got further (webcamoid didn't crash on start) by uninstalling the web cam driver and then reinstalling web cam driver manually. It still crashes when selecting an output source though.

KWB, 2022-06-10:
Same issue here:

I install _akvirtualcamera_ via the Webcamoid 9.0.0 settings ("The virtual camera is not installed, do you want to install it?"). This installs _akvirtualcamera_windows 9.1.0_. After completing and restarting _Webcamoid 9.0.0_ hangs forever.

Uninstalling _akvirtualcamera_windows_ solves the issue. I have also manually tried _akvirtualcamera_windows 9.0.0_ with the same result (Webcamoid hanging). Also as said above, the second installation from within Webcamoid 9.0.0  does not work, so I have to install manually.

Anything I can do to resolve the issue?

```
OS Name	Microsoft Windows 10 Pro
Version	10.0.19044 Build 19044
System Name	DELL-INSPIRON
System Model	Inspiron 7548
System Type	x64-based PC
```

Edit: Also have tested the daily build of Webcamoid, same issue. I use a non standard path **without** space for installation, `P:\Tools\AkVirtualCamera`

dtantono, 2022-07-20:
I confirm that it does not work on Windows 10 pro version 10.0.19044

Ben-Nachie, 2022-07-26:
Same issue here. Tried the installer and the portable versions, not the daily, and found that it crashes after the virtual camera driver is installed. I was unable to get it running again, even after uninstalling both Webcamoid and the driver.

Saya47, 2022-09-27:
I'm having the same issues!
Any workaround so far?

hipersayanX, 2022-09-28:
@Saya47 Try uninstalling AkVirtualCamera from Control Panel and [resetting the configurations](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh#windows).

ender1324, 2022-10-26:
same issue, it hangs after installing the virtual camera driver on windows 11

krillep, 2023-01-16:
Well.. 11 months later and it still goes into not responding endlessly after installing the virtual driver :/
(on new Windows 11 install)

hipersayanX, 2023-01-16:
@krillep try the [Daily build](https://github.com/webcamoid/webcamoid/releases/tag/daily-build), I have tried just a few minutes ago and it was working fine.

cjbarth, 2023-01-16:
I concur with @krillep . I just tried the daily build again, and, same behavior.

hipersayanX, 2023-01-16:
@cjbarth I downloaded the daily build, installed the virtual camera, tested a few minutes with Firefox, closed, opened again, tested, closed, and opened again, everything was working perfectly fine for me, I can't reproduce the behavior as described in the first post :confused:
Any missing information I need to know? maybe the virtual camera wasn't the problem?

KWB, 2023-01-16:
Just tried the x64 daily build on Windows 10. Issue persists:

- In about 80-90% of all cases the light of the camera switches on an the program hangs (not responding)
- In 1 or 2 startups it was not hanging, but in this case the camera light was off and did not seem to use the camera

cjbarth, 2023-01-17:
@hipersayanX , I'm not sure what to say. It is clear that I'm not the only one. I can grab some video footage if you'd like, but there really isn't much to show. When it hangs, I let it report the problem to Microsoft. I can create a dump file using Task Manager when the app is frozen if that would help you. I feel like I have a reasonably generic setup, so I'm not sure what kind of information would help you.

I'm involved in some other projects right now, so I can't re-setup the dev tools for this project to try to see where the hang is until May, at the earliest. Several people have posted to this thread, perhaps they can set up the dev tools and try to see if they can reproduce the hang.

hipersayanX, 2023-01-18:
There is a new [Daily Build](https://github.com/webcamoid/webcamoid/releases/tag/daily-build), I have fixed some bugs with the virtual camera, DirectShow and Media Foundation, maybe this fixed this bug?

@cjbarth 

>I'm not sure what to say. It is clear that I'm not the only one.

I know, I understand, the problem for me is that I can't reproduce the bug at all, maybe because my working environment is totally different from people having the bug, maybe because the steps to reproduce the bug is too vague for me to understand, or whatever.
And because of that, I haven't other option than doing blind debugging. It's similar to what happened with #148, in that case I was not able to reproduce the bug because the integrated camere I use to do quick test has not an auto exposure control, so the bug was never triggered, until someone mentioned that the problem was in that control.
In similar way, there is something in your environment that is not in mine.

>I'm involved in some other projects right now, so I can't re-setup the dev tools for this project to try to see where the hang is until May, at the earliest. Several people have posted to this thread, perhaps they can set up the dev tools and try to see if they can reproduce the hang.

This is what would help me a lot, is a lot much faster that the people having the bug just debug the program and send me the results (or the PR). Anyway, compiling Webcamoid in Windows is not that much hard, is just installing MSYS2, the dependencies and running cmake, no much special configuration is needed.

KWB, 2023-01-19:
I have used the latest build and the issue seems to be solved for me. It is actually the first time I could start the programming with camera working.

cjbarth, 2023-01-19:
> There is a new [Daily Build](https://github.com/webcamoid/webcamoid/releases/tag/daily-build), I have fixed some bugs with the virtual camera, DirectShow and Media Foundation, maybe this fixed this bug?

Thank you so much! Whatever change you made, this seems to have fixed it. I'll go about more detailed testing later, but I just wanted to give you the positive feedback :)

hipersayanX, 2023-01-20:
@KWB 

>Should those 2 not appear as cameras under the Windows devices? Sorry if I ask in the wrong place.

In fact, it should. If is as @cjbarth said [here](https://github.com/webcamoid/akvirtualcamera/issues/42#issuecomment-1397055004), maybe the plugin need some adjustment, anyway I'll release Webcamoid first and will check that issue later.

@cjbarth ready for a tale story? when posted [this](https://github.com/webcamoid/webcamoid/issues/497#issuecomment-1383986773) everything was working fine, then uninstalled the virtual camera (this is important later) and proceeded to [fix all code related to the frame reading](https://github.com/webcamoid/webcamoid/commit/9c39d68c3cb19bb1b9088bccdbabee288073e95e) because no picture were visible, screen capture was working fine (again, this is important), finished the work and proceed to test... baaam, neither the camera capture nor the desktop capture was working :sweat: 

[image]

Tracing the frames, there only 2 places they can go in the code, either the [video display](https://github.com/webcamoid/webcamoid/blob/3c0d1fead61bab363b537bb3fd29c10321c40898/StandAlone/src/videodisplay.cpp#L118) (the frames were not arriving there), and... the [virtual camera input](https://github.com/webcamoid/webcamoid/blob/3c0d1fead61bab363b537bb3fd29c10321c40898/libAvKys/Plugins/VirtualCamera/src/virtualcameraelement.cpp#L537), commented out the ```write()``` line, and everything was working again :smile: , the problem was that it was trying to write the frames even when the camera was not installed or working,  so fixed the problem, and continue.   

The other bug I had was that when closing Webcamoid, the process just keep running at infinity and beyond, and even killing Webcamoid, AkVCamManager still keep running as a zombie process, so traced the code again, and found that the process was stuck at [terminate()](https://github.com/webcamoid/webcamoid/blob/3c0d1fead61bab363b537bb3fd29c10321c40898/libAvKys/Plugins/VirtualCamera/src/dshow/src/vcamdshow.cpp#L1244) because... the process was still [reading the output](https://github.com/webcamoid/webcamoid/blob/3c0d1fead61bab363b537bb3fd29c10321c40898/libAvKys/Plugins/VirtualCamera/src/dshow/src/vcamdshow.cpp#L1244) of the manager even when it was supposedly terminated, so fixed it and just in case [killed the manager](https://github.com/webcamoid/webcamoid/blob/3c0d1fead61bab363b537bb3fd29c10321c40898/libAvKys/Plugins/VirtualCamera/src/dshow/src/vcamdshow.cpp#L1245), killing the process was probably unnecessary but just let me have the flavor of victory for a moment with a blood bath :smiling_imp: :laughing: 

Is not programming funny?

cjbarth, 2023-01-22:
> Is not programming funny?

It sure is. Thanks for your efforts though! I appreciate knowing what the problems was. Experience like that helps me to think of other places to look when there is problematic code. Keep up the good work 🙂

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX