webcamoid/webcamoid #498: 9.0.0: The Webcamoid screen just mirrors display

yurivict, 2022-02-22:
No UI or webcam is displayed.

Is this normal?

FreeBSD 13

yurivict, 2022-02-22:
It also prints a lot of these messages:
```
[2022-02-22 00:23:02.530, Webcamoid, 0x805e12000, OptionList.qml (57)] warning: qrc:/Webcamoid/share/qml/OptionList.qml:57: TypeError: Cannot read property 'left' of null
[2022-02-22 00:23:02.530, Webcamoid, 0x805e12000, OptionList.qml (56)] warning: qrc:/Webcamoid/share/qml/OptionList.qml:56: TypeError: Cannot read property 'right' of null
[2022-02-22 00:23:02.530, Webcamoid, 0x805e12000, OptionList.qml (57)] warning: qrc:/Webcamoid/share/qml/OptionList.qml:57: TypeError: Cannot read property 'left' of null
[2022-02-22 00:23:02.530, Webcamoid, 0x805e12000, OptionList.qml (56)] warning: qrc:/Webcamoid/share/qml/OptionList.qml:56: TypeError: Cannot read property 'right' of null
[2022-02-22 00:23:02.530, Webcamoid, 0x805e12000, OptionList.qml (57)] warning: qrc:/Webcamoid/share/qml/OptionList.qml:57: TypeError: Cannot read property 'left' of null
[2022-02-22 00:23:02.530, Webcamoid, 0x805e12000, OptionList.qml (56)] warning: qrc:/Webcamoid/share/qml/OptionList.qml:56: TypeError: Cannot read property 'right' of null
[2022-02-22 00:23:02.530, Webcamoid, 0x805e12000, OptionList.qml (57)] warning: qrc:/Webcamoid/share/qml/OptionList.qml:57: TypeError: Cannot read property 'left' of null
[2022-02-22 00:23:02.530, Webcamoid, 0x805e12000, OptionList.qml (56)] warning: qrc:/Webcamoid/share/qml/OptionList.qml:56: TypeError: Cannot read property 'right' of null
[2022-02-22 00:23:02.530, Webcamoid, 0x805e12000, OptionList.qml (57)] warning: qrc:/Webcamoid/share/qml/OptionList.qml:57: TypeError: Cannot read property 'left' of null
[2022-02-22 00:23:02.530, Webcamoid, 0x805e12000, OptionList.qml (56)] warning: qrc:/Webcamoid/share/qml/OptionList.qml:56: TypeError: Cannot read property 'right' of null
[2022-02-22 00:23:02.530, Webcamoid, 0x805e12000, OptionList.qml (57)] warning: qrc:/Webcamoid/share/qml/OptionList.qml:57: TypeError: Cannot read property 'left' of null
```

Qt-5.15.2
kf5-kdeclarative-5.89.0

hipersayanX, 2022-02-22:
There is not any other messages before or after those messages? is webcamoid process using the CPU?

yurivict, 2022-02-22:
All messages are here:
```
$ webcamoid 
[2022-02-22 12:31:34.860, webcamoid, 0x805e12000,  (0)] warning: Qt: Session management error: Authentication Rejected, reason : None of the authentication protocols specified are supported and host-based authentication failed
W: [(null)] caps.c: Normally all extra capabilities would be dropped now, but that's impossible because PulseAudio was built without capabilities support.
W: [(null)] caps.c: Normally all extra capabilities would be dropped now, but that's impossible because PulseAudio was built without capabilities support.
[2022-02-22 12:31:36.644, Webcamoid, 0x805e12000,  (0)] debug: Added component "Webcamoid"
[2022-02-22 12:31:37.942, Webcamoid, 0x805e12000, VideoOutputPicture.qml (73)] warning: qrc:/Webcamoid/share/qml/VideoOutputPicture.qml:73:13: QML Image: Protocol "" is unknown
```

Malachai80, 2022-05-11:
I had exactly the same issue, but I found the solution (at least for me) in the screenshots of the UI. The mirroring of my desktop was so fast and chaotic, that I just overlooked the little gear for the settings in the right upper corner of the window. Once I clicked it and changed the source from screen to my webcam everything was fine.

hipersayanX, 2023-04-19:
Could this be the same as  #586? because at the end it seems this was solved.

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX