webcamoid/webcamoid #499: Can't run on 10.14.6 (requires 10.11 or later)

hevisko, 2022-02-23:
Just now downloaded 9.0.0 and the installed as well as portable option gives this error pop-up :(

[image]

hipersayanX, 2022-02-25:
Weird, I been tested it in Catalina and it was working fine.

canalnoises, 2022-06-27:
I just downloaded 9.0.0 myself and have this same problem on 10.14.6. Looks like testing it in Catalina isn't enough ;)

[image]

gdinwiddie, 2022-07-01:
I have this problem, also.

hipersayanX, 2022-09-28:
Not sure, mut maybe [this PR](https://github.com/webcamoid/webcamoid/pull/564) could have fixed this bug? (did not tested yet).
@canalnoises I will keep supporting this old version of Mac as long as I can, I have not money to buy a Mac, and even if I have the money I prefer to save it, if no developer step up to support Mac I will remove support soon or later.

canalnoises, 2022-09-30:
Oh, awesome. So if I build from source now, it should include that PR, right? I'll give that a try and test it out.

Totally understand about Mac support. We'll enjoy it and appreciate it for however long you can support it :)

hipersayanX, 2022-09-30:
@canalnoises Is dying [somewhere in here](https://github.com/webcamoid/webcamoid/blob/6093d8cc68592f3e35c68c7a1812d89ba96a9f95/libAvKys/Plugins/VideoCapture/src/capture/avfoundation/src/captureavfoundation.mm#L611), probably [here](https://github.com/webcamoid/webcamoid/blob/6093d8cc68592f3e35c68c7a1812d89ba96a9f95/libAvKys/Plugins/VideoCapture/src/capture/avfoundation/src/captureavfoundation.mm#L683), because Apple has the marvelous idea of crashing apps (on pourpose) in which the user did not give explicit permissions for accessing the resources.

https://developer.apple.com/documentation/avfoundation/avcapturedevice/1624584-requestaccessformediatype

>Your app must provide an explanation for its use of capture devices using the [NSCameraUsageDescription](https://developer.apple.com/documentation/bundleresources/information_property_list/nscamerausagedescription) and [NSMicrophoneUsageDescription](https://developer.apple.com/documentation/bundleresources/information_property_list/nsmicrophoneusagedescription) Info.plist keys. The system presents the strings you set for these keys when prompting the user for permission, and thereafter in the Settings app. **Calling this method or attempting to start a capture session without a usage description raises an exception**.

canalnoises, 2022-09-30:
It asked for camera permissions the first time I launched the app, and I verified that Webcamoid is showing as authorized to access the camera in the Security & Privacy System Preferences. Unfortunately it also crashes when I try to launch the binary directly.

hipersayanX, 2022-09-30:
@canalnoises just in case, do you have experience programming, working with Cmake, homebrew, Xcode or Qt Creator? so we can advance quickly.

hipersayanX, 2023-04-27:
This won't be possible to fix, Github only provide [macOS 11 Big Sur images and higher](https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners#supported-runners-and-hardware-resources), but you can still [build it from scratch](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Mac).

closed 2023-04-27 by hipersayanX