webcamoid/webcamoid #500: Getting current Appimage to run on Linux Mint

mdevour, 2022-02-24:
I downloaded the x86_64 Appimage for GNU/Linux from the project main page, made the file executable, and attempted to run it in terminal from the download directory. Similar to issue #480, the terminal output reports over a dozen 'not found' elements.

The following terminal command produces this output:

mike@mymachine:~/Downloads/Webcamoid$ ./webcamoid-portable-linux-9.0.0-x86_64.AppImage

```
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_webcamM3m1kH/bin/../lib/libQt5Widgets.so.5)
webcamoid: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /tmp/.mount_webcamM3m1kH/bin/../lib/libQt5Widgets.so.5)
webcamoid: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.28' not found (required by /tmp/.mount_webcamM3m1kH/bin/../lib/libQt5Widgets.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_webcamM3m1kH/bin/../lib/libQt5Quick.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_webcamM3m1kH/bin/../lib/libQt5Qml.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_webcamM3m1kH/bin/../lib/libQt5Gui.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_webcamM3m1kH/bin/../lib/libQt5Core.so.5)
webcamoid: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/.mount_webcamM3m1kH/bin/../lib/libQt5Core.so.5)
webcamoid: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /tmp/.mount_webcamM3m1kH/bin/../lib/libQt5Core.so.5)
webcamoid: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.28' not found (required by /tmp/.mount_webcamM3m1kH/bin/../lib/libQt5Core.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_webcamM3m1kH/bin/../lib/./libpng16.so.16)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_webcamM3m1kH/bin/../lib/./libicui18n.so.66)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_webcamM3m1kH/bin/../lib/./libicuuc.so.66)
```

I have searched and do not have a ~/.config/Webcamoid folder (or anything similar or anywhere else) to delete to reset the configuration.

Is the appimage packaged properly?

I recognize this is a recurrent problem. I've read through the similar Issues in hopes of finding a fix. I'm looking forward to being able to run the program, and especially using the virtual camera capability.

My environment:

Webcamoid:
> webcamoid-portable-linux-9.0.0-x86_64.AppImage

System:   
>    Distro: Linux Mint 19.3 Tricia 
>    Base: Ubuntu 18.04 bionic  
>    Kernel: 5.4.0-99-generic x86_64
>    Compiler: gcc v: 7.5.0 
>    Desktop: MATE 1.22.2  

Camera:
>    Logitech Brio 4k Pro Webcam (which works otherwise)

hipersayanX, 2022-02-25:
It's stated in the [website](http://webcamoid.github.io/#downloads-gnu-linux), you need glibc 2.31 and gcc 9.3 or higher. The AppImage and all binaries are built in Ubuntu Focal Fossa, you need [Linux Mint Ulyana](https://linuxmint.com/download_all.php) or higher, you must  upgrade your system.

mdevour, 2022-02-25:
Thank you for your reply. 

I believed that an AppImage would rely on its own resources entirely, by definition. I did not understand that it would be possible or necessary for it to have external dependencies? I thought that would only apply to compiling source or installable packages.

I will add this issue to the list of reasons to upgrade my system. When there are enough and vital enough I will do so. 

Thank you for the project. I look forward to trying the program when I do upgrade.

hipersayanX, 2022-03-03:
>I believed that an AppImage would rely on its own resources entirely, by definition.

Yes and no, the software packaged with AppImage will only work in distros of the same version or newer than the distro you used to compile it, it's explained in the [AppImage documentation](https://docs.appimage.org/reference/best-practices.html#binaries-compiled-on-old-enough-base-system).
Webcamoid binaries are compiled in Focal Fossa, you can't use in older distros than that.

mdevour, 2022-03-13:
Thank you for explaining. As I read the referenced docs, I now understand that the developer's choice of environment determines the oldest distro the appimage will run on. The more bleeding edge the development system, the fewer users will be able to run it. Conversely, you reach a point of diminishing returns depending on an older system to keep working, as I am presently. ;)
I'll close this issue. Thanks!

closed 2022-03-13 by mdevour