webcamoid/webcamoid #502: Add basic example

jidanni, 2022-03-10:
If
$ cheese
puts my webcam image on my screen,
then please document how to do it with webcaminod.

jidanni, 2022-03-10:
Tons more must be  missing too:
[image]

hipersayanX, 2022-03-10:
>Tons more must be missing too:

Just open a different issue for that.

barak, 2022-03-11:
There's already tons of others there.
Anyway, I added this one to the Debian package, just uploaded. Let me know if you notice anything else missing...

hipersayanX, 2022-03-11:
@barak 

>if you notice anything else missing...

It should be Debian users  who must report missing dependencies, take in mind that I'm giving support to [13 platforms](https://github.com/webcamoid/webcamoid/tree/master/.github/workflows) ([including FreeBSD](https://cirrus-ci.com/github/webcamoid/webcamoid)) and I can't test all of them.

>Also, shouldn't there be a list of these somewhere? I used ports/ci/linux-debian/install_deps.sh but if that's not complete...

I don't know if the dependencies are complete or not, the dependencies are added based on the dependencies needed to run the [AppImage](https://github.com/webcamoid/webcamoid/runs/5503128585?check_suite_focus=true#step:5:10692) more or less. I won't create separated list of dependencies for every single platform because it would be unmaintainable.

If more developers were collaborating with the project, that would be a different story...

barak, 2022-03-12:
@hipersayanX I'm not asking for Debian-specific help. In fact, something that helps keep all distributions working would be best. Maybe there is some generic system-independent way of listing all the qt-module-like dependencies? Like maybe just module name as qt views it, or something like that. Then it would be up to individual distributions or systems to set up their own tooling to convert such a list into a concrete list of packages, or whatever.

hipersayanX, 2022-03-12:
@Maybe 

>there is some generic system-independent way of listing all the qt-module-like dependencies?

A solution could be writing the build dependencies to a file when running the pipelines and then uploading those files somewhere, something like:

_install_deps.sh_

```
# Dependencies list.
deps="
    gstreamer1.0-plugins-base \
    gstreamer1.0-plugins-good \
    libasound2-dev \
    libavcodec-dev \
    ...

# Write dependencies list to a file.
cat << EOF > debian_dependencies.txt
${deps}
EOF
"

# Upload the dependencies list.
curl -F ‘data=@debian_dependencies.txt’ somewhere

# Continue with the rest of the script.
apt-get -y install \
    ${deps}
```

the problem is then, where and how?

jidanni, 2022-03-13:
By the way, back to the very original problem, it would be good to document that when one starts the program, once you do these steps, then  we'll see one's webcam image on one's screen. At least that way people will see if what they see is what is correct or not. If not that means they haven't installed the program correctly.

iam-TJ, 2022-05-31:
I've started on my idea for a user manual in my forked repository's Wiki. Please [take a look](https://github.com/iam-TJ/webcamoid/wiki/Webcamoid---getting-started) and confirm if this style is acceptable, in which case I'll continue developing it.

hipersayanX, 2022-05-31:
@iam-TJ looks good, In sent you an invitation for editing the wiki.

iam-TJ, 2022-05-31:
Thanks - I'll continue developing it on my own Wiki for now (I generally iterate many times until I'm happy with it) then can transfer it across once it is stable.

jidanni, 2022-06-02:
Thanks, but I'm not using my web camera anymore, so must leave...