webcamoid/webcamoid #505: Please add to F-Droid.

BEEDELLROKEJULIANLOCKHART, 2022-03-27:
# Want a new feature?

"https://github.com/webcamoid/webcamoid/releases/download/daily-build/webcamoid-android-daily-master-1.apk" demonstrates that the code currently compiles to Android. I realise that the current state of this software for Android is objectvely inadequate, as "http://github.com/webcamoid/webcamoid/issues/169#issuecomment-1013526028" demonstrates, but addition of it to F-Droid despite that would allow users of it that potentially do not want to compile this to test it, and would allow me to automate my process of installation of it preparatively.

I am thankful for any consideration.

hipersayanX, 2022-03-28:
But F-Droid has some way of marking an app as development? because the problem here is that Webcamoid for Android can't be considered as stable at all, I've tested it and its quite far of being stable or usable.

BEEDELLROKEJULIANLOCKHART, 2022-03-28:
It is able to mark it as developmental.

http://user-images.githubusercontent.com/42837531/160484997-98ef482a-92c5-469c-b600-987551639544.png

BEEDELLROKEJULIANLOCKHART, 2022-03-28:
...However, note that the sole problem that I have encountered that was significantly problematic was the lack of adaptibility of the configurative interface.

hipersayanX, 2022-03-30:
Will check then, thanks!

hipersayanX, 2022-04-04:
>...However, note that the sole problem that I have encountered that was significantly problematic was the lack of adaptibility of the configurative interface.

If that were the only problem...
I was testing it and there are much more problems that just that

- The app is crashing when screen capturing.
- The app hangs when using a hi resolution video source and entering to video effects dialog (as in #488).
- Can't scroll the effects, [Qml ScrollView](https://doc.qt.io/qt-5/qml-qtquick-controls2-scrollview.html) does not work in Android.

BEEDELLROKEJULIANLOCKHART, 2022-04-10:
However, despite all of those problems, I have successfully utilised Webcamoid as my sole mobile camera application since I posted my request, and am adequately satisfied by merely being able to utilise the same software across my devices.

hipersayanX, 2023-06-24:
I have tried everything that was on my hand, but it's impossible to comply with F-Droid rules.
[This commit](https://gitlab.com/webcamoid/fdroiddata/-/pipelines/908794729) was the nearest I was of building Webcamoid with their rules, but [lint step fails](https://gitlab.com/webcamoid/fdroiddata/-/jobs/4524391670) with no clear indication of the error.
Then tried [building Qt from sources](https://gitlab.com/webcamoid/fdroiddata/-/jobs/4534707585) (maybe that can calm the error above?) but failed because it can't find find g++ for whatever reason.
For me it's totally discouraging having to depend on third-parties rules for distributing my software, if somebody is interested on maintaining it for me its' ok, otherwise the official packages for Android can be downloaded [here](https://github.com/webcamoid/webcamoid/releases/tag/9.1.1).

rokejulianlockhart, 2023-06-25:
@hipersayanX, have you considered just providing your own F-Droid repository? That's what Collabora and Bitwarden do.

hipersayanX, 2023-06-25:
@rokejulianlockhart and where I'm supposed to take the money to pay the server? I'm can give you free features for free in my free time, don't expect nothing more nothing else, and [I'm currently receiving 0$/month](https://github.com/sponsors/hipersayanX) with a very few (counted with one hand) donations per year that does not even count to pay rents, so this is the best you can expect.
For the same reason is that why I've removed MacOs packages and Flathub links, the more complicated they do, the less interest I've on maintaining it, but of course I haven't not problem if people [put some effort in their side](https://github.com/webcamoid/webcamoid/issues/592).

rokejulianlockhart, 2023-06-25:
@IzzySoft, would you consider adding this?

rokejulianlockhart, 2023-06-25:
@hipersayanX, [I see that you have tried to commit some work](https://gitlab.com/webcamoid/fdroiddata/-/pipelines/908794729), but have you opened a standard packaging request at F-Droid's GitLab repository? Usually some people are able to help there. If you haven't already, I suggest that you read https://f-droid.org/en/docs/Inclusion_How-To/.

IzzySoft, 2023-06-29:
> would you consider adding this to your repository (in the meantime)?

I certainly would, but the APK is more than 3 times the per-app limit in my repo, sorry :cry:

hipersayanX, 2023-07-10:
@rokejulianlockhart As I understand, I must build everything from source code, including Qt, so I've added Qt to the build instead of using [aqtinstall](https://github.com/miurahr/aqtinstall) with pre-compiled Qt as I was doing before, and now is throwing this error.

```
Creating qmake...
gmake: g++: No such file or directory
gmake: *** [Makefile:250: main.o] Error 127
```

It's probably some simple line I've to modify out there, but right now and most probably for the next months I won't have much time and energy to put into the project. I don't have time for asking in forums, issue trackers, waiting for an answer, apply the "solution", watch it fail, and repeat the loop again.
That's why I prefer this issue to be taken by another person, and leave for me the most essential parts.