webcamoid/webcamoid #506: Please support RAW.

BEEDELLROKEJULIANLOCKHART, 2022-04-01:
# Want a new feature?

Please support the "http://wikipedia.org/wiki/Raw_image_format" because the format allows retainment of approximately all of the information that was provided to what created the photograph, as "http://adobe.com/creativecloud/file-types/image/raw.html" states.

I suggest that implementation of this should utilise "http://helpx.adobe.com/camera-raw/digital-negative.html" so that the information is simultaneously standardised and extensible so that parsal and archival are more easy than they alternatively would be.

# Questions and other matters?

Approximately all professional software supports this format, which even includes software that is designed for mobile devices, as "http://reddit.com/comments/4uh4v5 demonstrates", so I believe that the lack of this ability within software that is as impressive as Webcamoid is weirdly unusual.

I have not been able to locate how to activate this ability if Webcamoid supports it, potentially because I have solely evaluated "http://github.com/webcamoid/webcamoid/releases/download/daily-build/webcamoid-android-daily-master-1.apk", so I apologise if this is unnecessary.

hipersayanX, 2022-04-02:
Sure, anyway Webcamoid capture the camera images as given by the camera, then it transform to other formats for applying the video effects and recording, I don't know if I need to do something else for supporting that, do it need some special image/video saving format? what if the user want to apply effects to the image? how the program should go into one or another mode? how the program should interact in that case?

BEEDELLROKEJULIANLOCKHART, 2022-04-09:
@hipersayanX, my proposition is to support a format, which I believe that what I have cited demonstrates.

A RAW file is simply a digital image file that is stored on your camera or smartphones memory card. It is minimally processed and is usually uncompressed.

Every camera manufacturer has their own RAW file format, for example Canon RAW files are .CR2 or .CR3, whilst Nikon are .NEF.

When it comes to smartphones, most Android smartphones which support RAW primarily shoot in DNG, which is a universal RAW file format.

There are downsides to using a compressed file like JPG or HEIF. A compressed image is a lot less flexible when it comes to editing, as much of the useful image data that image editors work with has been discarded in order to save file size. Additionally, the camera applies a number of tweaks to the image when it saves it in a compressed format, including adjusting the saturation, contrast, and sharpness of the image. These changes are difficult to undo if you don’t like them.

A RAW file on the other hand, is an uncompressed version of the image file. Essentially the camera takes the image data from the sensor, and saves it in an unedited and uncompressed format on the memory card.