webcamoid/webcamoid #507: Dialog will only open on Screen #0

svanschalkwyk, 2022-04-06:
5.13.0-39-generic #44-Ubuntu SMP Thu Mar 24 15:35:05 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 21.04
Mate Desktop
Webcamoid Ver 9.0.0

Dialog can not be moved from Display 0. 
Dragging always immediately reverts to screen 0.

hipersayanX, 2022-04-07:
It could be related to #117, but honestly don't expect it to be fixed, it looks more like a bug in Qt code rater than a bug in Webcamoid.

jvaccaro, 2022-04-13:
Same happens on Ubuntu 20.04 LTS w X windowing

MrAwesome, 2022-05-04:
Encountering this on Arch Linux using Awesome WM. Webcamoid absolutely refuses to be on Screen 1 - it will force itself back to Screen 0 no matter what happens. And oddly enough, the Screen 1 input seems to always be garbled, regardless of which screen is actually designated Screen 1. So just by moving screens to the left, they appear correctly in the stream, but webcamoid pins itself to that screen hard. This is obviously an issue when trying to stream, as I'd like webcamoid to not record itself if possible. As things stand, it's always going to do so since Screen 1 is garbled.

I'd love to help debug this further, if possible. @hipersayanX do you have any idea where I could report this upstream?

hipersayanX, 2022-05-05:
@MrAwesome all Qt bugs can be reported [here](https://bugreports.qt.io/). Also, just in case, the only code controlling the window position is [this](https://github.com/webcamoid/webcamoid/blob/7c68a5b1f117c32467fff092d8c38eca4aa4a34f/StandAlone/share/qml/main.qml#L37):

```
ApplicationWindow {
    // ...
    x: (Screen.width - mediaTools.windowWidth) / 2
    y: (Screen.height - mediaTools.windowHeight) / 2
    width: mediaTools.windowWidth
    height: mediaTools.windowHeight
    // ...
}
```

**mediaTools.windowWidth** and **mediaTools.windowHeight** can be taken as constants, therefore I don't have any control on which screen the window should be opened.

iam-TJ, 2022-05-30:
Following up from issue #513, I added some debug code to report position changes to get some feedback on why this happens. With two screens (monitors) side-by-side, 1920x1200 (screen 0) -> 1600x1200 (screen 1, KDE 'primary') and `webcamoid` always starting on screen 0, first relocation the application window further and further to the right of screen 0 until more than half of it is on screen 1 - at which point the window 'snaps' back to center of screen 0. Then attempt to move it completely onto screen 1. As soon as the the mouse button is released the window snaps back to screen 0.
```
$ /usr/local/bin/webcamoid 
[2022-05-30 20:05:56.563, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 480 y -> 348 width = 640 height = 504
[2022-05-30 20:05:56.563, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 480 y = 348 width = 640 height = 504
[2022-05-30 20:05:57.182, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 640 y = 348 width = 640 height = 504
[2022-05-30 20:05:57.243, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 640 y = 348 width = 640 height = 504
[2022-05-30 20:05:57.244, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 640 y = 348 width = 640 height = 504
[2022-05-30 20:05:57.819, Webcamoid, 0x7f2056ffd700,  (0)] debug: Camera input frame format: AkCaps(mimeType="video/unknown",fourcc=QVariant(QString, "MJPG"),fps=QVariant(QString, "30/1"),height=QVariant(uint, 1080),width=QVariant(uint, 1920))
[2022-05-30 20:06:03.717, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 718 y = 497 width = 640 height = 504
[2022-05-30 20:06:03.717, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 718 y -> 497 width = 640 height = 504
[2022-05-30 20:06:07.252, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 1280 y = 541 width = 640 height = 504
[2022-05-30 20:06:07.252, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 1280 y -> 541 width = 640 height = 504
[2022-05-30 20:06:11.685, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 1632 y = 537 width = 640 height = 504
[2022-05-30 20:06:11.685, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 1632 y -> 537 width = 640 height = 504
[2022-05-30 20:06:11.686, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 480 y = 348 width = 640 height = 504
[2022-05-30 20:06:11.686, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 480 y -> 348 width = 640 height = 504
[2022-05-30 20:06:11.686, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 640 y = 348 width = 640 height = 504
[2022-05-30 20:06:11.686, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 480 y = 537 width = 640 height = 504
[2022-05-30 20:06:11.686, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 480 y -> 537 width = 640 height = 504
[2022-05-30 20:06:11.687, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 480 y -> 348 width = 640 height = 504
[2022-05-30 20:06:11.687, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 640 y = 348 width = 640 height = 504
[2022-05-30 20:06:15.100, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 1517 y = 554 width = 640 height = 504
[2022-05-30 20:06:15.100, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 1517 y -> 554 width = 640 height = 504
[2022-05-30 20:06:16.772, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 1544 y = 553 width = 640 height = 504
[2022-05-30 20:06:16.772, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 1544 y -> 553 width = 640 height = 504
[2022-05-30 20:06:21.500, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 1559 y = 551 width = 640 height = 504
[2022-05-30 20:06:21.500, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 1559 y -> 551 width = 640 height = 504
[2022-05-30 20:06:23.893, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 1608 y = 554 width = 640 height = 504
[2022-05-30 20:06:23.893, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 1608 y -> 554 width = 640 height = 504
[2022-05-30 20:06:23.893, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 480 y = 348 width = 640 height = 504
[2022-05-30 20:06:23.893, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 480 y -> 348 width = 640 height = 504
[2022-05-30 20:06:23.894, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 640 y = 348 width = 640 height = 504
[2022-05-30 20:06:23.895, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 480 y = 554 width = 640 height = 504
[2022-05-30 20:06:23.895, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 480 y -> 554 width = 640 height = 504
[2022-05-30 20:06:23.895, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 480 y -> 348 width = 640 height = 504
[2022-05-30 20:06:23.895, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 640 y = 348 width = 640 height = 504
[2022-05-30 20:06:29.180, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 2833 y = 433 width = 640 height = 504
[2022-05-30 20:06:29.180, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 2833 y -> 433 width = 640 height = 504
[2022-05-30 20:06:29.181, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 480 y = 348 width = 640 height = 504
[2022-05-30 20:06:29.181, Webcamoid, 0x7f212854e7c0, main.qml (77)] debug: x = 480 y -> 348 width = 640 height = 504
[2022-05-30 20:06:29.181, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 640 y = 348 width = 640 height = 504
[2022-05-30 20:06:29.181, Webcamoid, 0x7f212854e7c0, main.qml (76)] debug: x -> 480 y = 433 width = 640 height = 504
[... cut ...]
diff --git a/StandAlone/share/qml/main.qml b/StandAlone/share/qml/main.qml
index 344a0d0a..5ba9553b 100644
--- a/StandAlone/share/qml/main.qml
+++ b/StandAlone/share/qml/main.qml
@@ -73,6 +73,8 @@ ApplicationWindow {
 
     onWidthChanged: mediaTools.windowWidth = width
     onHeightChanged: mediaTools.windowHeight = height
+    onXChanged: console.debug("x ->",x,"y =",y,"width =",width,"height =",height)
+    onYChanged: console.debug("x =",x,"y ->",y,"width =",width,"height =",height)
 
     Component.onCompleted: chkFlash.updateVisibility()

```

iam-TJ, 2022-05-30:
OK, another one solved. The patch below fixes this. I'll generate a PR later once I've synced my local repository with my github fork!
```
$ git status -u no; git show HEAD
On branch issue507-allow-positioning-on-any-screen
nothing to commit, working tree clean
commit 3a1a26a8a3098ec114172c5d279be9cbf5ed635f (HEAD -> issue507-allow-positioning-on-any-screen)
Author: Tj <hacker@iam.tj>
Date:   Mon May 30 21:00:48 2022 +0100

    ApplicationWindow: allow positioning on any screen
    
    Properties x and y were incorrectly bound such that any time the window
    position changed it was reset. The intention was to only set the initial
    window position when the application starts.
    
    Move the code to the onCompleted() signal so it only fires once.
    
    Closes issue #507.

diff --git a/StandAlone/share/qml/main.qml b/StandAlone/share/qml/main.qml
index 65a8c5b2..d764e9fe 100644
--- a/StandAlone/share/qml/main.qml
+++ b/StandAlone/share/qml/main.qml
@@ -34,8 +34,6 @@ ApplicationWindow {
            + " - "
            + videoLayer.description(videoLayer.videoInput)
     visible: true
-    x: (Screen.width - mediaTools.windowWidth) / 2
-    y: (Screen.height - mediaTools.windowHeight) / 2
     width: mediaTools.windowWidth
     height: mediaTools.windowHeight
 
@@ -74,7 +72,11 @@ ApplicationWindow {
     onWidthChanged: mediaTools.windowWidth = width
     onHeightChanged: mediaTools.windowHeight = height
 
-    Component.onCompleted: chkFlash.updateVisibility()
+    Component.onCompleted: {
+        x = (Screen.width - mediaTools.windowWidth) / 2
+        y = (Screen.height - mediaTools.windowHeight) / 2
+        chkFlash.updateVisibility()
+    }
 
     Connections {
         target: mediaTools
```

hipersayanX, 2022-05-31:
The weird things of Qml, I assumed that the window position was set only on startup.

iam-TJ, 2022-06-03:
This can be closed now.

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX