webcamoid/webcamoid #509: Version 9.0.0 still saving 0-byte MP4 files

svanschalkwyk, 2022-04-06:
5.13.0-39-generic #44-Ubuntu SMP Thu Mar 24 15:35:05 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 21.04
Mate Desktop
Webcamoid Ver 9.0.0

Still saving 0-byte MP4 files.

svanschalkwyk, 2022-04-06:
Errors on install: 
```
000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/codec/libttml_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/codec/libzvbi_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/codec/libomxil_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/codec/libspdif_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/codec/libtextst_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/codec/libaraw_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/codec/libkate_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/codec/libx265_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/codec/libfaad_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/codec/libshine_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/codec/libopus_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/codec/libsubsdec_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/audio_mixer/libfloat_mixer_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/audio_mixer/libinteger_mixer_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/services_discovery/libxcb_apps_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/services_discovery/libpulselist_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/services_discovery/libmediadirs_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/services_discovery/libmtp_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/services_discovery/libupnp_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/services_discovery/libpodcast_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/services_discovery/libavahi_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/services_discovery/libsap_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/services_discovery/libudev_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/gui/libncurses_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/audio_output/libadummy_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/audio_output/libafile_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/audio_output/libpulse_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/audio_output/libalsa_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/audio_output/libsndio_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/audio_output/libamem_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/spu/liblogo_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/spu/libmarq_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/spu/libremoteosd_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/spu/librss_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/spu/libdynamicoverlay_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/spu/libmosaic_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/spu/libaudiobargraph_v_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/spu/libsubsdelay_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/video_output/libyuv_plugin.so
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/video_output/libvmem_plugin.so
[... cut ...]
[000055820ebde0f0] main libvlc error: stale plugins cache: modified /usr/share/webcamoid/lib/vlc/plugins/meta_engine/libfolder_plugin.so
[000055820eca93a0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[000055820eca93a0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[2022-04-06 18:49:54.080, Webcamoid, 0x7f5d837fe640,  (0)] debug: Camera input frame format: AkCaps(mimeType="video/unknown",fourcc=QVariant(QString, "YUYV"),fps=QVariant(QString, "30/1"),height=QVariant(uint, 480),width=QVariant(uint, 640))

```

hipersayanX, 2022-04-07:
Ignore libvlc errors, those does not affect Webcamoid. Is there any error after that? any error appearing after clicking the recording button?
Just by chance (by pure chance) I have  found the following error: 

```
[libx264 @ 0x5590d5ea3240] height not divisible by 2 (632x617)
[2022-04-06 22:43:44.142, Webcamoid, 0x7f383b4aed00,  (0)] debug: Error:  Generic error in an external library
[mp4 @ 0x5590d5ea2740] Could not find tag for codec none in stream #0, codec not currently supported in container
[2022-04-06 22:43:44.205, Webcamoid, 0x7f383b4aed00,  (0)] debug: Can't write header:  Error number 22 occurred
```

But I don't know if this your case or not. Recording in the same format as you (YUYV 640x480 30 FPS) is working fine for me, anything special you are doing? What format, audio and video codecs are you using?

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX