webcamoid/webcamoid #510: webcamoid 9.0.0 cant add virtual camera using pkexec

mixalbl4-127, 2022-04-13:
> [2022-04-13 11:03:36.017, Webcamoid, 0x7f974b9f27c0,  (0)] debug: pkexec --version |
>        --help |
>        --disable-internal-agent |
>        [--user username] PROGRAM [ARGUMENTS...]
> 
> See the pkexec manual page for more details.

App: webcamoid-portable-linux-9.0.0-x86_64
OS: Ubuntu 20.04

previous version (8+) works fine with it.
fount same error: https://www.mail-archive.com/debian-bugs-dist@lists.debian.org/msg1845604.html

hipersayanX, 2022-04-13:
Does this command works for you?

```
echo "echo Hello world" | pkexec
```

mixalbl4-127, 2022-04-18:
@hipersayanX 
`$ echo "echo Hello world" | pkexec`
```
pkexec --version |
       --help |
       --disable-internal-agent |
       [--user username] PROGRAM [ARGUMENTS...]

See the pkexec manual page for more details.

```

hipersayanX, 2022-04-19:
Ok, passing the commands through the stdin works perfectly fine in Arch, doesn't work in Debian derivatives, I can then write the commands to a script fine, but then Debian users will be exposed to a [security risk](https://github.com/webcamoid/webcamoid/issues/330), I can then maybe write a command just for updating the virtual camera settings and call it with pkexec, but don't expect it until maybe hopefully next year because there are no other developers working on the project. What should I do then? :angry:

billypurdue, 2022-04-21:
> 

In the meantime is there a way we can create the virtual camera by hand?  I was poking around in the source but I'm not nearly smart enough to figure that out.  Otherwise for many this will be useless until it's resolved.

hipersayanX, 2022-04-22:
@billypurdue yes, just check the [wiki](https://github.com/webcamoid/akvcam/wiki).

mixalbl4-127, 2022-04-23:
```
echo "echo Hello world" | pkexec /bin/sh
Hello world
```
works fine! @hipersayanX

hipersayanX, 2022-04-24:
You can suggest Debian team to patch [these](https://github.com/webcamoid/webcamoid/blob/3bf5c1d99a0f406be5de84a9d8bddd8ca66f4a80/libAvKys/Plugins/VirtualCamera/src/akvcam/src/vcamak.cpp#L1966) [lines](https://github.com/webcamoid/webcamoid/blob/3bf5c1d99a0f406be5de84a9d8bddd8ca66f4a80/libAvKys/Plugins/VirtualCamera/src/v4l2lb/src/vcamv4l2lb.cpp#L1204) to

```
su.start(sudoBin, QStringList {"/bin/sh"});
```

hipersayanX, 2022-06-03:
@iam-TJ I usually prefer distros to follow upstream, but looking at your explanation, they are actually right. Anyway it was not a big deal to fix that.

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX