webcamoid/webcamoid #511: Feature Request: Be able to auto rotate video

broes5, 2022-04-13:
# Feature Request
To have video auto rotate when the orientation of the laptop is changed, or have a setting in "Video Source Options" that can be used to manually change the video orientation.

## Reason
I own a ThinkPad L13 Yoga Gen 2. It is a convertible laptop with a second webcam mounted next to the keyboard, which is intended to be used when the laptop is in a tablet configuration. When I flip the laptop all the way around, so that the keyboard is facing in the opposite direction as the screen and then change the video source to the keyboard webcam, the video is upside down. I know that the laptop has an accelerometer, and the desktop environment is able to auto rotate the display when I flip my laptop into the tablet configuration. 

I'm aware of the matrix transform effect would be able to invert the video, but it's intuitive and is applied to all video sources, which in my is case is not necessary for the first webcam (the one next to the screen). I'm not sure that there are many others in the same situation as me.

BEEDELLROKEJULIANLOCKHART, 2022-05-04:
This affects Android too.

broes5, 2022-05-05:
How so?

BEEDELLROKEJULIANLOCKHART, 2022-05-08:
@broes5, when I rotate my device, all of the interface of Webcamoid except the viewfinder of it rotates accordantly. This forces my device to remain horizontal when I utilise Webcamoid.

iam-TJ, 2022-06-01:
Auto-rotation of the application and camera view works correctly on current git HEAD when on GNU/Linux (Debian 11 Bullseye, amd64, kernel v5.18, amdgpu, KDE with two monitors). I tested using `xrandr --output DisplayPort-1 --rotate {left|right|normal}` and the camera view rotated same as the application and desktop.
```
$ /usr/local/bin/webcamoid 
[2022-06-01 12:44:09.022, Webcamoid, 0x7f065fcc47c0,  (0)] debug: Screen count: 2
[2022-06-01 12:44:09.022, Webcamoid, 0x7f065fcc47c0,  (0)] debug: Primary: "DisplayPort-0"
[2022-06-01 12:44:09.022, Webcamoid, 0x7f065fcc47c0,  (0)] debug: Tracking screen: "DisplayPort-0"
[2022-06-01 12:44:09.022, Webcamoid, 0x7f065fcc47c0,  (0)] debug: Tracking screen: "DisplayPort-1"
[2022-06-01 12:44:10.994, Webcamoid, 0x7f05923fe700,  (0)] debug: Camera input frame format: AkCaps(mimeType="video/unknown",fourcc=QVariant(QString, "MJPG"),fps=QVariant(QString, "30/1"),height=QVariant(uint, 1080),width=QVariant(uint, 1920))
[2022-06-01 12:44:42.870, Webcamoid, 0x7f065fcc47c0,  (0)] debug: Orientation changed for screen "DisplayPort-1" : Qt::PortraitOrientation
[2022-06-01 12:45:37.433, Webcamoid, 0x7f065fcc47c0,  (0)] debug: Orientation changed for screen "DisplayPort-1" : Qt::InvertedPortraitOrientation
[2022-06-01 12:45:44.829, Webcamoid, 0x7f065fcc47c0,  (0)] debug: Orientation changed for screen "DisplayPort-1" : Qt::LandscapeOrientation
```

broes5, 2022-06-04:
It doesn't seem to work for me. I tested it using the same command on an Arch laptop with Intel graphics hardware.

iam-TJ, 2022-06-04:
@broes5 For the avoidance of doubt, is it correct that your system is using X not a Wayland compositor, and you used the output names unique to your system as shown by `xrandr -q` ?

Also, which desktop environment (and version) is in use (KDE, Gnome, etc.) ?

If that is the case please run these commands and copy/paste them and their results from within the GUI session:
```
lspci -nnk -d ::0300
lspci -nnk -d ::0320
```
And in each orientation of the tablet run this command `xrandr -q`, post the results and tell us the orientation for each result.

broes5, 2022-06-05:
Yes, @iam-TJ my system is using X and I did use a different output name (`eDP-1`) to the one in your example. I have KDE Plasma 5.24.5 installed, which I'm pretty sure is the most recent version. The second command didn't seem to produce any output. Here's the output of the first command:

```
00:02.0 VGA compatible controller [0300]: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] [8086:9a49] (rev 01)
        Subsystem: Lenovo Device [17aa:5089]
        Kernel driver in use: i915
        Kernel modules: i915
```

Here is the output of `xrandr -q` in each screen orientation:

Normal:
```
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
eDP-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 293mm x 165mm
   1920x1080     60.03*+  60.01    59.97    59.96    59.93  
   1680x1050     59.95    59.88  
   1400x1050     59.98  
   1600x900      59.99    59.94    59.95    59.82  
   1280x1024     60.02  
   1400x900      59.96    59.88  
   1280x960      60.00  
   1440x810      60.00    59.97  
   1368x768      59.88    59.85  
   1280x800      59.99    59.97    59.81    59.91  
   1280x720      60.00    59.99    59.86    59.74  
   1024x768      60.04    60.00  
   960x720       60.00  
   928x696       60.05  
   896x672       60.01  
   1024x576      59.95    59.96    59.90    59.82  
   960x600       59.93    60.00  
   960x540       59.96    59.99    59.63    59.82  
   800x600       60.00    60.32    56.25  
   840x525       60.01    59.88  
   864x486       59.92    59.57  
   700x525       59.98  
   800x450       59.95    59.82  
   640x512       60.02  
   700x450       59.96    59.88  
   640x480       60.00    59.94  
   720x405       59.51    58.99  
   684x384       59.88    59.85  
   640x400       59.88    59.98  
   640x360       59.86    59.83    59.84    59.32  
   512x384       60.00  
   512x288       60.00    59.92  
   480x270       59.63    59.82  
   400x300       60.32    56.34  
   432x243       59.92    59.57  
   320x240       60.05  
   360x202       59.51    59.13  
   320x180       59.84    59.32  
HDMI-1 disconnected (normal left inverted right x axis y axis)
DP-1 disconnected (normal left inverted right x axis y axis)
DP-2 disconnected (normal left inverted right x axis y axis)
DP-3 disconnected (normal left inverted right x axis y axis)
DP-4 disconnected (normal left inverted right x axis y axis)
```

Inverted:
```
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
eDP-1 connected primary 1920x1080+0+0 inverted (normal left inverted right x axis y axis) 293mm x 165mm
   1920x1080     60.03*+  60.01    59.97    59.96    59.93  
   1680x1050     59.95    59.88  
   1400x1050     59.98  
   1600x900      59.99    59.94    59.95    59.82  
   1280x1024     60.02  
   1400x900      59.96    59.88  
[... 109 lines cut ...]
   840x525       60.01    59.88  
   864x486       59.92    59.57  
   700x525       59.98  
   800x450       59.95    59.82  
   640x512       60.02  
   700x450       59.96    59.88  
   640x480       60.00    59.94  
   720x405       59.51    58.99  
   684x384       59.88    59.85  
   640x400       59.88    59.98  
   640x360       59.86    59.83    59.84    59.32  
   512x384       60.00  
   512x288       60.00    59.92  
   480x270       59.63    59.82  
   400x300       60.32    56.34  
   432x243       59.92    59.57  
   320x240       60.05  
   360x202       59.51    59.13  
   320x180       59.84    59.32  
HDMI-1 disconnected (normal left inverted right x axis y axis)
DP-1 disconnected (normal left inverted right x axis y axis)
DP-2 disconnected (normal left inverted right x axis y axis)
DP-3 disconnected (normal left inverted right x axis y axis)
DP-4 disconnected (normal left inverted right x axis y axis)
```

iam-TJ, 2022-06-05:
@broes5 Thank-you for that - it is very helpful. `xrandr` results show the system definitely knows the different rotations so this is puzzling!

I tested on my desktop development system. I do have a similar Asus transformer tablet/notebook hybrid (T300CHI) which also auto-detects rotations so I'll install `webcamoid` on that and see if I can reproduce your experience.