webcamoid/webcamoid #513: Segfault when primary screen unplugged

nahoj, 2022-04-23:
## Summary

On a multi-screen setup with Webcamoid open, if I unplug the screen that was primary when I opened the app, the app crashes. This happens even if the app's window is not on the unplugged screen.

I also have issues #507 and #117: The app opens always in the same place at the top of my laptop screen (which is not the primary screen), partly out of the screen. I can drag it and resize it freely on the laptop screen but if I move it to the other screen, it snaps back to where it opened.

## Current Behavior

```
[2022-04-23 13:29:54.418, Webcamoid, 0x7fbeddda57c0, VideoEffectsPanel.qml (56)] warning: qrc:/Webcamoid/share/qml/VideoEffectsPanel.qml:56:9: QML VideoEffectOptions: Binding loop detected for property "implicitWidth"
Segmentation fault (core dumped)
```

Note: about 1 second elapses between the VideoEffectsPanel warning and the segfault. Upon other changes to the multi-screen configuration, I get the VideoEffectsPanel warning without the segfault.

## Expected Behavior

No crash. 

## Steps to Reproduce

1.  Have a multi-monitor setup. Know which monitor is the primary.
2.  Open Webcamoid
3.  Unplug the primary monitor (the one that was primary when Webcamoid started)

## Suggestions and tips

Could there be a mixup between attaching the app to the actual screen it's on vs. to the virtual screen that contains all the screens + blank spaces?

In #507 and #117, Qt is named as a potential culprit, but as a KDE user I have tons of Qt apps and this is the only one for which this happens.

## Your Environment

* Webcamoid information: 9.0.0 x86_64
  * I tried to reproduce with Webcamoid 8.6.1 and I think it crashed once but I can't reproduce it reliably.
* Operating System information: Kubuntu 21.10 x86_64 (Linux 5.13.0-39-generic, X, KDE, XRender)
* Hardware: Laptop + external monitor, Intel + nVidia GPUs (nVidia drivers with "On Demand" PRIME profile).

hipersayanX, 2022-04-24:
>Qt is named as a potential culprit, but as a KDE user I have tons of Qt apps and this is the only one for which this happens.

Most KDE apps are written using QtWidgets and their subclasses, Webcamoid is written in Qml, based on my experience with it, Qml is a pile of bugs.

iam-TJ, 2022-05-29:
I'll test this since I use multiple monitors with KDE on Debian 11 Bullseye (AMD Ryzen 7 + AMD FirePro 4x DisplayPort).

iam-TJ, 2022-05-30:
I've tried all manner of layout and un-plug scenarios and cannot reproduce the segfault with Debian 11 Bullseye, kernel v5.18, amdgpu, KDE, AMD Ryzen 7 + AMD FirePro GPU with 2 monitors attached.

As you mention nVidia and Prime that suggests you're using the proprietary nVidia driver so possibly something to examine. Check the kernel, Xorg and `~/.xsession-errors` logs when the unplug occurs for clues.

In respect of the application snapping back to a monitor (KDE screen) I can reproduce that when the application is not maximised. Very annoying because there are some _sneaky_ ways to manage it with judicious grab/move procedures!

nahoj, 2022-05-30:
Here are the logs when the bug occurs:

**kern.log**

```
May 30 19:18:38 <myhostname> kernel: [673659.213942] traps: webcamoid[330959] general protection fault ip:7f57decd5da4 sp:7ffd248c93f8 error:0 in libQt5Gui.so.5.15.2[7f57dec52000+4d4000]
```

**Xorg.0.log**

```
[290449.977] (II) modeset(0): EDID vendor "NCP", prod id 80
[290449.977] (II) modeset(0): Using hsync ranges from config file
[290449.977] (II) modeset(0): Using vrefresh ranges from config file
[290449.977] (II) modeset(0): Printing DDC gathered Modelines:
[290449.977] (II) modeset(0): Modeline "1920x1080"x0.0  335.38  1920 1968 2000 2180  1080 1083 1088 1282 +hsync -vsync (153.8 kHz eP)
[290449.977] (II) modeset(0): Modeline "1920x1080"x0.0  167.69  1920 1968 2000 2180  1080 1083 1088 1282 +hsync -vsync (76.9 kHz e)
[290451.478] (II) modeset(0): Allocate new frame buffer 1920x1080 stride
```

<details>
  <summary>~/.xsession-errors</summary>

```
(check_displays                ) Displays changed!
(check_displays                ) Previous connected displays: card0-DP-1, card0-eDP-1
(check_displays                ) Current  connected displays: card0-eDP-1
(dummy_display_change_handler  ) changes = Changed_Removed
(dummy_display_change_handler  ) Removed displays: card0-DP-1
Old primary output: QScreen(0x7fd96c04e4f0, name="DP-1") New primary output: QScreen(0x7fd96c018cc0, name="eDP-1")
trying to show an empty dialog
trying to show an empty dialog
trying to show an empty dialog
trying to show an empty dialog
trying to show an empty dialog
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Desktop.qml:146:19: QML Loader: Binding loop detected for property "height"
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Desktop.qml:146:19: QML Loader: Binding loop detected for property "height"
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Panel.qml:103: TypeError: Cannot read property 'enabledBorders' of null
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Panel.qml:95: TypeError: Cannot read property 'enabledBorders' of null
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Panel.qml:80: TypeError: Cannot read property 'screenGeometry' of null
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Panel.qml:167: TypeError: Cannot read property 'opacityMode' of null
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 62625, resource id: 88087442, major code: 15 (QueryTree), minor code: 0
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 62923, resource id: 88087470, major code: 15 (QueryTree), minor code: 0
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 63057, resource id: 88087494, major code: 18 (ChangeProperty), minor code: 0
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Desktop.qml:121: TypeError: Cannot read property 'x' of null
file:///usr/share/plasma/shells/org.kde.plasma.desktop/contents/views/Desktop.qml:135: TypeError: Cannot read property 'y' of null
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 63314, resource id: 88087492, major code: 15 (QueryTree), minor code: 0
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 30403, resource id: 88087492, major code: 3 (GetWindowAttributes), minor code: 0
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 30404, resource id: 88087492, major code: 2 (ChangeWindowAttributes), minor code: 0
Qt: Session management error: networkIdsList argument is NULL
kf.kirigami: Units.devicePixelRatio is deprecated (since 5.86 ): This returns 1 when using Qt HiDPI scaling.
kf.kirigami: Previous message repeats 7 times.
file:///usr/share/kpackage/kcms/kcm_kscreen/contents/ui/OutputPanel.qml:126:9: QML SpinBox: Binding loop detected for property "implicitWidth"
file:///usr/share/kpackage/kcms/kcm_kscreen/contents/ui/OutputPanel.qml:126:9: QML SpinBox: Binding loop detected for property "implicitWidth"
file:///usr/share/kpackage/kcms/kcm_kscreen/contents/ui/OutputPanel.qml:126:9: QML SpinBox: Binding loop detected for property "implicitHeight"
file:///usr/share/kpackage/kcms/kcm_kscreen/contents/ui/OutputPanel.qml:74:13: QML SpinBox: Binding loop detected for property "implicitWidth"
file:///usr/share/kpackage/kcms/kcm_kscreen/contents/ui/OutputPanel.qml:74:13: QML SpinBox: Binding loop detected for property "implicitWidth"
file:///usr/share/kpackage/kcms/kcm_kscreen/contents/ui/OutputPanel.qml:74:13: QML SpinBox: Binding loop detected for property "implicitHeight"
file:///usr/share/kpackage/kcms/kcm_kscreen/contents/ui/OutputPanel.qml:126:9: QML SpinBox: Binding loop detected for property "implicitWidth"
file:///usr/share/kpackage/kcms/kcm_kscreen/contents/ui/OutputPanel.qml:74:13: QML SpinBox: Binding loop detected for property "implicitWidth"
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 6035, resource id: 92274694, major code: 15 (QueryTree), minor code: 0
```

</details>

hipersayanX, 2022-05-30:
@nahoj Just in case, try deleting the following files (or were they are supposed to be installed):

```
webcamoid/lib/avkys/libDesktopCapture_pipewire.so
webcamoid/lib/avkys/libDesktopCapture_qtscreen.so
```

and see if it still crash.

iam-TJ, 2022-05-30:
The earlier mention that the application reports
```
... QML VideoEffectOptions: Binding loop detected for property "implicitWidth"
```
makes me wonder if this is only happening when the video effects option dialog for some effect is displayed? My tests are with the default source output on start-up with no effects applied.

iam-TJ, 2022-05-30:
Continuing investigating the snapping-back-to-initial-screen (monitor) issue #507

nahoj, 2022-05-30:
> @nahoj Just in case, try deleting (or moving outside the directory tree) the following files (or were they are supposed to be installed):
> 
> ```
> webcamoid/lib/avkys/libDesktopCapture_pipewire.so
> webcamoid/lib/avkys/libDesktopCapture_qtscreen.so
> ```
> 
> and see if it still crash.

It doesn't! Everything seems to be working fine with this fix except that the screens are not available as source. Thanks!

> makes me wonder if this is only happening when the video effects option dialog for some effect is displayed? My tests are with the default source output on start-up with no effects applied.
> 
> Can you also confirm the input source is a camera and not a screen?

The crash happens right after opening Webcamoid, with a webcam as source, the effects dialog closed and no effect in use.

(And with the 2 libs removed, no crash even while in the effects settings.)

hipersayanX, 2022-05-31:
Then the problem should be somewhere in [here](https://github.com/webcamoid/webcamoid/blob/f25b8b1c2d01d5ad3222804638e5b9dfae8ad672/libAvKys/Plugins/DesktopCapture/src/qtscreen/src/qtscreendev.cpp#L338) or [here](https://github.com/webcamoid/webcamoid/blob/f25b8b1c2d01d5ad3222804638e5b9dfae8ad672/libAvKys/Plugins/DesktopCapture/src/qtscreen/src/qtscreendev.cpp#L100).

hipersayanX, 2023-04-28:
I've simplified the code from the [screen devices](https://github.com/webcamoid/webcamoid/blob/da7a3efc39278d8f2a569c6c21157a921265532b/libAvKys/Plugins/DesktopCapture/src/qtscreen/src/qtscreendev.cpp#L316) update. Somebody can test it?

hipersayanX, 2023-06-10:
It should be working fine in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX