webcamoid/webcamoid #515: Please enable Discussion.

BEEDELLROKEJULIANLOCKHART, 2022-05-03:
# Questions and other matters?

As I believe that "http://github.com/webcamoid/webcamoid/issues/81" demonstrates, "http://docs.github.com/discussions" is useful for this repository.

hipersayanX, 2022-05-03:
[Enabled](https://github.com/webcamoid/webcamoid/discussions).

closed 2022-05-03 by hipersayanX

hipersayanX, 2025-06-21:
TBH I didn't remember I had disabled it 😆 . There were not too much participation at all really, and I don't see any special reason to have 2 different spaces for discussing things about the project. Using it as forum? maybe, but basically any discussion can turn into an issue that can be better classified, and now days forums are pretty much dead, when most people are in chatting social networks like X and others. I can open it again but 🤷‍♂

RokeJulianLockhart, 2025-06-21:
> I don't see any special reason to have 2 different spaces for discussing things about the project. Using it as forum? maybe, but basically any discussion can turn into an issue that can be better classified

@hipersayanX, you're probably right. I think GitHub would have been better-off introducing a "Question" tag to all new repositories by default, and/or implementing basic threading, like GitLab has.