webcamoid/webcamoid #521: [Constructive Criticism] General project problem

qkum, 2022-05-05:
A wiki page + webpage that focus on everything else than telling the user how to use the software.

No "quick start" section, or guidance to how you add your camera/webcam if it does not pop up by itself?

I for example use a raspberry pi camera _Raspberry Pi High Quality Camera_ https://www.raspberrypi.com/products/raspberry-pi-high-quality-camera/.

My version of webcamoid is 8.5.0 installed from Raspberry Pi OS arm64 install apps, app.

I wont wait for a answer but find another solution. but now you know where your focus should be short term, and no other place,  than fixing this basic beginner section of any serious, widely used, Open Source project.

Dont make it a headache/pain to be a user, please.

qkum, 2022-05-05:
Having the basics of making a serious tutorial on how to use your software is mandatory before you ask for donations/sponsors....this is so off the rails that I cant fathom it in all seriousness....I'm disappointed.

qkum, 2022-05-05:
And how can anybody make an Arm64 version, or even Arm32 version without plug and play support for the Original Raspberry Pi Cameras? 

I hope I can shut up now, even though my disappointment on the behalf of the Open Source community, is as big as the ocean right now.

hipersayanX, 2022-05-05:
The general project problem is the lack of software developers et all, being me the only developer that must carry with all the weight of: developing Webcamoid, the virtual cameras, solving issue, answering questing, documenting things, for free without even receiving a fucking thanks, and charring a life outside the project.
It would be much easier if users starts contributing instead complaining.

>I hope I can shut up now, even though my disappointment on the behalf of the Open Source community, is as big as the ocean right now.

The point of Free Software is to share knowledge and helping each other for common good, not my problem if you are lazy to do so :kissing_heart:

iam-TJ, 2022-06-01:
I'm addressing this via issue #502.