webcamoid/webcamoid #523: Cannot create virtual camera in webcamoid 9.0.0

nosfirlabs, 2022-05-27:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debugger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates below that fit better your issue.

# Reporting a problem?

## Summary

After setting op the input sources, going to "Outputs" GUI, clicking Add output opens up a popup window with a list of codecs, resolutions and FPS, i click on one and it opens another Window called "Change Video Format", by clicking ok it closes again and back to the list of codecs. User is not able to create a virtual webcam, by clicking OK. WIndow closes and nothing happens.

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue below:
V4L2 with wrapper library
Qt screen capture
PulseAudio
FFMpeg
VLC
FFMpeg
akvcam
pkexec

## Your Environment

* Webcamoid information: 9.0.0 x64
* Operating System information: Ubuntu 20.04.4 LTS 64-Bit

hipersayanX, 2022-05-29:
This issue could be related to #510.

nosfirlabs, 2022-07-18:
Anyone has a fix?

hipersayanX, 2023-06-10:
Bug probably fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX