webcamoid/webcamoid #524: 9.0.0 Virtual camera - "Installation failed - execvp: Permission denied"

iam-TJ, 2022-05-28:
## Summary

I've built from the current git head (`f25b8b1c`) today. Went to add a virtual camera and that unexpectedly triggers an automatic 25MB download rather than using the installed `v4l2loopback` (which is used by 8.6.1 (Debian Bullseye)). The downloaded file is written to `/tmp/` but it is not possible to execute it since `/tmp/` is a `tmpfs` mount and has `noexec` mount option (`tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noexec,relatime,mode=777)`) - a common configuration to prevent malicious code execution.

Thus an error dialog reports `Installation failed` ` execvp: Permission denied`

## Expected Behavior

It might be better to download to a directory below the user home directory, possibly `$HOME/.cache/Webcamoid/`, since that is already used.

Also note that, on a 1920x1080 monitor, with the application maximized, the Download dialog and the error dialog both truncate the title and text and the dialogs are not resizable so important information is unreadable.

v9.0.0 (commit `f25b8b1c`) amd64
Debian Bullseye (kernel 5.18)

iam-TJ, 2022-05-29:
Subsequent investigation (see my other related issue reports) shows this was due to not building with the dependency `libkmod-dev.

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX