webcamoid/webcamoid #525: 9.0.0 Corrupted first entry in plugin list

iam-TJ, 2022-05-28:
## Summary

`Configure > Plugins > Plugins` list has its first entry corrupted - looks to be over-written with tick marks. When hovering pointer over the entry it appears correctly but as soon as focus moves elsewhere the corrupted graphics returns. This happens on repeated launches of the program. Happens even after resizing the window.

v9.0.0 (commit `f25b8b1c1`) amd64
Debian 11 Bullseye, KDE
[image]

iam-TJ, 2022-05-28:
This looks like it was caused by some left-over configuration from the Debian Bullseye v8.6.1 installation. After `rm -r $HOME/.config/Webcamoid/*` and restarting the application it, so far, appears to be behaving correctly now.

hipersayanX, 2022-05-29:
I can't reproduce the bug, it's probably a bug with Qml itself or your graphics card, I don't control nothing related to that area, so don't expect this bug to be fixed.

closed 2022-05-29 by iam-TJ