webcamoid/webcamoid #527: 9.0.0 - some configuration operations cause excess CPU and eats memory

iam-TJ, 2022-05-28:
## Summary

When opening the left-side filter list I noticed a frequent lock-up during a visual fade transition to the list, combined with excessive CPU usage and the process rapidly eating memory. It appears to be frequently reproducible. I also encountered the same issue when trying to configure a virtual camera output - application hung and I noticed memory usage rapidly passing 32% (of 32GiB) with 52% CPU usage (on 8 cores) - had to `pkill` the process before it caused OOM problems.

v9.0.0 (commit `f25b8b1c`) amd64
Debian 11 Bullseye, KDE

iam-TJ, 2022-05-29:
I've tracked memory usage in the following circumstances and this shows it being eaten from the moment the filter preview begins:

1. Start program windowed (not maximised) with auto-start of source (see attached screenshot) from terminal shell
2. Run a simple `ps` monitor in a terminal shell at 11:40:52
3. Choose `Filter` icon, then `Add effect`
4. Application creates list of effects and defaults to first in list `ASCII Art` and appears to process first video frame
5. Subsequently the preview doesn't appear to update as the camera view changes
6. At log time 11:41:15 I clicked on `Cartoon` effect name
7. No changes to the preview
8. At 11:41:50 I clicked on `Cancel` - no response
9. At 11:42:00 I interrupt the process in the shell with `Ctrl+C`

Commands in the terminal shells are:
```
$ webcamoid --log-file /tmp/webcamoid.log
[2022-05-29 11:40:46.291, Webcamoid, 0x7fd97314e7c0,  (0)] debug: Sending log to "/tmp/webcamoid.log"
```
and to monitor:
```
$ echo -en "Time \t"; ps -fly -p ${RANDOM}${NOEXIST}; while true; do  echo -en $(date +%H:%M:%S); ps --no-headers -fly -p $(pgrep webcamoid); sleep 1; done |& tee /tmp/webcamoid-track-memory.log
```
Screenshot and logs attached.
[image]
[webcamoid.log](https://github.com/webcamoid/webcamoid/files/8793162/webcamoid.log)
[webcamoid-track-memory.log](https://github.com/webcamoid/webcamoid/files/8793167/webcamoid-track-memory.log)

iam-TJ, 2022-05-29:
Further investigation reveals at least part of the issue appears to be the size of the camera input. The camera `HD USB PTZ` is configured for 1920x1080@30fps. Reducing that to 640x480@30fps results in the `Effects` list being slightly more responsive and renders some previews. There are still significant hangs/delays in response which become more pronounced the longer preview runs and it still eats memory.

As this doesn't affect 8.6.1 it would seem to be some kind of regression.

iam-TJ, 2022-05-29:
This may be related to issue #488 although it is unclear what these "Natural" and "Fixed" modes are and how to configure them.

iam-TJ, 2022-05-29:
After some manually driven git-bisect operations between tags v8.6.1 and v9.0.0, and ignoring the pre-cmake build era (because the old qmake/Makefile wasn't setting the install prefix to `/usr/local/`) I've so far reached the described commit where the application appears to behave correctly with the camera configured to 1920x1080@30fps, although some effects do show frame lag compared to v8.6.1 (e.g. cartoon, firey):
```
2d578378 2021-04-20 19:33:51 -0300 E Gonzalo Exequiel Pedone Added missing dependency.
```

iam-TJ, 2022-05-29:
Completed the bisect - unfortunately the problem seems to be in a a commit with "several big improvements" !
```
$ git bisect log
git bisect start
# good: [2d578378e274dd48a7e7602cee11093eea2b1389] Added missing dependency.
git bisect good 2d578378e274dd48a7e7602cee11093eea2b1389
# bad: [2cb2f47fc2245a630bdd425818b2abe8a2801b10] Merge branch 'master' of github.com:webcamoid/webcamoid
git bisect bad 2cb2f47fc2245a630bdd425818b2abe8a2801b10
# good: [53252a0a2113ae505e2a1ad2161dc03856e77105] Fixing actions 26.
git bisect good 53252a0a2113ae505e2a1ad2161dc03856e77105
# good: [53252a0a2113ae505e2a1ad2161dc03856e77105] Fixing actions 26.
git bisect good 53252a0a2113ae505e2a1ad2161dc03856e77105
# good: [660001d21f539dfe2de106caffaef88673eaaaf5] Added missing dependency in fedora.
git bisect good 660001d21f539dfe2de106caffaef88673eaaaf5
# good: [0157473b6c96b946cbf082a7bf0f290e108b05d7] Translated using Weblate (German)
git bisect good 0157473b6c96b946cbf082a7bf0f290e108b05d7
# bad: [b2d60761387a730576bc347d98b78c8efd37744c] Trying to fix Fedora builds 3.
git bisect bad b2d60761387a730576bc347d98b78c8efd37744c
# bad: [3f8ba805140b820f1a69338c00632434ee8e694f] This is a commit with several big improvements.
git bisect bad 3f8ba805140b820f1a69338c00632434ee8e694f
# good: [35c030658053a858261a69715bc94d245d15d402] Merge branch 'master' of github.com:webcamoid/webcamoid
git bisect good 35c030658053a858261a69715bc94d245d15d402
```
The offending commit is a big one so I'm not sure where to look - is there a private branch containing these changes as separate commits that could be further bisected?
```
commit 3f8ba805140b820f1a69338c00632434ee8e694f (refs/bisect/bad)
Author: Gonzalo Exequiel Pedone <hipersayan.x@gmail.com>
Date:   Sun Sep 5 17:13:03 2021 -0300

    This is a commit with several big improvements.
    
    - Now plugins can be detected and registered independently of it's
    filename.
    - Submodules are now just another normal plugin, and you can use
    wildcard and priority rules to decide which one will be loaded.
    - Improved updates system, now it will be possible to update any
    Webcamoid component (for example the virtual camera).
    - Added a rudimentary download manager, for downloading Webcamoid
    components in future versions.
    - Added the code for downloading and installing the virtual camera (this
    feature isn't mapped to the UI yet).
```

iam-TJ, 2022-05-29:
Digging deeper my first suspicion is focused on `VideoEffects::setEffects()` and `VideoEffects::setState()` in `StandAlone/src/videoeffects.cpp`  - both of which contain interesting `for` loops that could explain the observed behaviour whilst setting the effect on the preview.

The `diff` shows wholesale changes (`@@ -1,756 +1,805 @@`) to the entire file making it harder to understand the changes.

hipersayanX, 2022-05-29:
This is the #488 for sure.

>This may be related to issue #488 although it is unclear what these "Natural" and "Fixed" modes are and how to configure them.

**Natural** it just draw the characters with the color taken from the camera, **Fixed** ignore the original colors and draw the characters with the foreground color. 

>Update: Discovered that effects cannot be configured until they have been added, and then their options are shown when choosing them in the Effects list itself.

Yes, I haven't found a good place to put the effect options in the preview. It's just a design matter.

>Also discovered the above "Natural" and "Fixed" are related only to the ASCII art effect.

Every effect has it's own options.

>This doesn't solve the original described issue since other effects exhibit the same problems - for example Cartoonify as one obvious example.

Cartoonify is also another resource hungry effect.

>The offending commit is a big one so I'm not sure where to look - is there a private branch containing these changes as separate commits that could be further bisected?

Those changes has absolutely nothing to do with effects lags and there is not any reason to be reverted, and won't be reverted. If an effect is running slow with higher resolution, then the code should be optimized more, or it should be detected when it's running at a slower frame rate and stop it. **Reverting code is never a solution for anything** (well, unless some university students want to introduce security bugs in your code).

Also, I understand that most developers recommends to split changes in small pieces, but that not the case when you have to rewrite the code, bigger problems requires bigger changes to the code, you can't just commit a few lines when the problem actually requires 10k lines or more to be modified, for the program to be properly working.

iam-TJ, 2022-05-29:
As the `git-bisect` log shows, the problem is contained within the single commit `3f8ba80514`. I've now extracted the before and after versions of `Standalone/src/videoeffects.cpp` as separate files so I can further understand those changes and experiment to determine if they may be responsible with some debugging aids in the current code.

Regarding commit sizes - the recommended and generally accepted approach (speaking now as a kernel developer) is to develop the changes on a feature branch as a series of (much) smaller commits. Once the feature is complete then it can be squashed into a single commit during a merge but the history is retained in the feature branch. That is what I was referring to when I mentioned a "private branch".

iam-TJ, 2022-05-29:
Using [meld](https://meldmerge.org/), the visual diff tool, on the isolated changes to `StandAlone/src/videoeffects.cpp`, is making it much easier to see that the changes to the file are actually quite easy to understand - the cause of the patch doing a wholesale delete/replace of all lines is all the blank lines that were removed.

In case others want to review the changes I'm attaching the before (`videoeffects.a.cpp`) and after (`videoeffects.b.cpp`) from commit `3f8ba80514`. Note that github won't upload `.cpp` files as attachments so I've added `.txt` to their names. Viewing these with `meld` is recommended.
[image]
[videoeffects.a.cpp.txt](https://github.com/webcamoid/webcamoid/files/8794392/videoeffects.a.cpp.txt)
[videoeffects.b.cpp.txt](https://github.com/webcamoid/webcamoid/files/8794393/videoeffects.b.cpp.txt)

hipersayanX, 2022-05-30:
And as I said before, those changes has nothing to do on how slow or fast the effects runs, those changes just change the way on how the effects are created or linked to the preview.

iam-TJ, 2022-05-30:
I know you said that - but the `git bisect` points to it.
```
# bad: [3f8ba805140b820f1a69338c00632434ee8e694f] This is a commit with several big improvements.
git bisect bad 3f8ba805140b820f1a69338c00632434ee8e694f
# good: [35c030658053a858261a69715bc94d245d15d402] Merge branch 'master' of github.com:webcamoid/webcamoid
git bisect good 35c030658053a858261a69715bc94d245d15d402
```
The commit logs around those is:
```
b81bf1fe 2021-09-05 17:13:33 -0300 E Gonzalo Exequiel Pedone Merge branch 'master' of github.com:webcamoid/webcamoid
**3f8ba805** 2021-09-05 17:13:03 -0300 E Gonzalo Exequiel Pedone This is a commit with several big improvements.
0157473b 2021-09-05 14:35:10 +0200 E nautilusx Translated using Weblate (German)
7bd59dad 2021-08-17 18:35:31 +0200 E Luna Jernberg Translated using Weblate (Swedish)
5623fee6 2021-08-15 14:32:55 +0200 E Artem Translated using Weblate (Russian)
ea5b84b3 2021-08-06 16:33:18 +0200 E lefiston Translated using Weblate (French)
d76ded62 2021-08-04 16:50:35 -0300 E Gonzalo Exequiel Pedone Contributors updated.
**35c03065** 2021-08-04 11:39:35 -0300 E Gonzalo Exequiel Pedone Merge branch 'master' of github.com:webcamoid/webcamoid
1d41258b 2021-08-04 11:39:23 -0300 E Gonzalo Exequiel Pedone Improved CI build scripts 12.
c1cce5f8 2021-08-02 17:34:23 +0200 E XiaoPanPanKevinPan Translated using Weblate (Chinese (Traditional))
```
We can ignore the merges commits themselves, and I'd be surprised if it is caused by a translation or contributors update.

I'll redo the `git bisect` surrounding those to confirm my earlier results.

hipersayanX, 2023-05-21:
I had tested with latest commit ([c4c3015](https://github.com/webcamoid/webcamoid/commit/c4c3015e591966928d2987753e325868bec05bc9)), played a 4K video and tested with **ASCII art** and  **Cartoonify**, and is resource hungry as expected, yes, with CPU at 55% and 1.5 Gb RAM usage, but memory usage is stable (not growing), and the UI is responsive, tested in a ```Intel(R) Core(TM) i3-3120M CPU @ 2.50GHz```. The virtual camera panel does not hung up any more.
So, the only problem remaining here is #488.

hipersayanX, 2023-06-10:
>When opening the left-side filter list I noticed a frequent lock-up during a visual fade transition to the list, c

Bug not reproducible, the other parts will require further redesign of the code, closing for now.

closed 2023-06-10 by hipersayanX