webcamoid/webcamoid #528: 9.0.0 - Virtual camera format width/height do not match parent entry

iam-TJ, 2022-05-28:
## Summary

When adding a virtual camera via `Video > Outputs > Add output` , in the `Add Virtual Camera` dialog selecting the first entry `yuyv422 640x480 30 FPS` results in a `Change Video Format` (`Remove format`) dialog with the `width = 215` and `Height = 26` - in fact this happens for all the listed formats I've tried ! 
[image]

## Expected Behavior

The dialog should show the correct parameters of the format about to be removed.

v9.0.0 (commit `f25b8b1c`) amd64
Debian 11 Bullseye, KDE

hipersayanX, 2022-05-29:
I've tried with both v4l2loopback and akvcam and it's working fine for me. The only bug I found right now was that selecting a 0rgb format, does not show it in the ComboBox.

iam-TJ, 2022-05-29:
I've just done a clean build of current HEAD to be sure, after deleting all `~/.cache/Webcamoid/` and `~/.config/Webcamoid/` files on top of
```
$ find /usr/local/ -iname '*webcamoid*' -o -name '*avk*' -o -wholename '*avkys/*' -delete
```
and the issue persists for both `akvcam` and `v4l2loopback` (originally I only tested with `akvcam`). I wondered if it were related to the window size when not maximised, or width/size of the configuration ribbon on the right, but those have no effect.

I'll add this to my todo-list to figure out.

iam-TJ, 2022-05-29:
After some `console.debug()` work in `StandAlone/share/qml/VideoOutputAddEdit.qml` I've tracked down the source of this error. The code is incorrectly using the attributes representing the size of the GUI element. It should be using `formatWidth` and `formatHeight` but is using `width` and `height`.
```
$ /usr/local/bin/webcamoid 
[2022-05-29 21:30:25.604, Webcamoid, 0x7f17b37fe700,  (0)] debug: Camera input frame format: AkCaps(mimeType="video/unknown",fourcc=QVariant(QString, "MJPG"),fps=QVariant(QString, "30/1"),height=QVariant(uint, 1080),width=QVariant(uint, 1920))
[2022-05-29 21:30:30.634, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (113)] debug: populateFormats() deviceId.text= 
[2022-05-29 21:30:30.635, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 148  resolution= 0  width= 640  height= 480
[2022-05-29 21:30:30.635, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 148  resolution= 1  width= 160  height= 120
[2022-05-29 21:30:30.635, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 148  resolution= 2  width= 320  height= 240
[2022-05-29 21:30:30.635, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 148  resolution= 3  width= 800  height= 600
[2022-05-29 21:30:30.635, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 148  resolution= 4  width= 1280  height= 720
[2022-05-29 21:30:30.635, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 148  resolution= 5  width= 1920  height= 1080
[2022-05-29 21:30:30.636, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 80  resolution= 0  width= 640  height= 480
[2022-05-29 21:30:30.636, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 80  resolution= 1  width= 160  height= 120
[2022-05-29 21:30:30.636, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 80  resolution= 2  width= 320  height= 240
[2022-05-29 21:30:30.636, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 80  resolution= 3  width= 800  height= 600
[2022-05-29 21:30:30.636, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 80  resolution= 4  width= 1280  height= 720
[2022-05-29 21:30:30.636, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 80  resolution= 5  width= 1920  height= 1080
[2022-05-29 21:30:30.636, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 65  resolution= 0  width= 640  height= 480
[2022-05-29 21:30:30.636, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 65  resolution= 1  width= 160  height= 120
[2022-05-29 21:30:30.636, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 65  resolution= 2  width= 320  height= 240
[2022-05-29 21:30:30.636, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 65  resolution= 3  width= 800  height= 600
[2022-05-29 21:30:30.636, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 65  resolution= 4  width= 1280  height= 720
[2022-05-29 21:30:30.636, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (143)] debug: Format= 65  resolution= 5  width= 1920  height= 1080
[2022-05-29 21:30:33.662, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (173)] debug: obj.onClicked.connect() index= 0  vcamFormats.itemAt[index]= VideoFormatItem_QMLTYPE_197(0x55ab34295f60)
[2022-05-29 21:30:33.663, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (175)] debug: caps from element:  148  width= 215 height= 26  fps= 30
[2022-05-29 21:30:33.663, Webcamoid, 0x7f188311f7c0, VideoOutputAddEdit.qml (176)] debug: caps from element:  148  formatWidth= 640 formatHeight= 480  fps= 30
```
Notice the last two lines - the first reports the attributes being used to populate the dialog - but those are the dimensions of the menu item as best as I can tell. The correct values are shown on the last line. Here's a simple patch to correct it. Do you want a PR for this?
```
$ git diff -- StandAlone/share/qml/VideoOutputAddEdit.qml 
diff --git a/StandAlone/share/qml/VideoOutputAddEdit.qml b/StandAlone/share/qml/VideoOutputAddEdit.qml
index 2c58b97b..a3a0f0a0 100644
--- a/StandAlone/share/qml/VideoOutputAddEdit.qml
+++ b/StandAlone/share/qml/VideoOutputAddEdit.qml
@@ -164,8 +164,8 @@ Dialog {
                 let element = vcamFormats.itemAt(index)
                 let caps =
                     AkVideoCaps.create(element.format,
-                                       element.width,
-                                       element.height,
+                                       element.formatWidth,
+                                       element.formatHeight,
                                        AkFrac.create(element.fps,
                                                      1).toVariant())
                 addEdit.openOutputFormatDialog(index, caps)
```
Fixed code now looks correct
[image]

hipersayanX, 2022-05-30:
Can you make a PR for this?

closed 2022-06-03 by iam-TJ