webcamoid/webcamoid #529: 9.0.0 -Truncated text for Photo-mode "Use Flash"

iam-TJ, 2022-05-29:
## Summary

Only just realised what the `Use ...` text in top-left corner beneath the `Effects` icon is supposed to be! In Photo mode it is supposed to show "Use Flash" and the list control below it contains delay times.

## Current Behavior

No matter what size the application window this text is always truncated (ellipsed) as `Use ...` - presumably due to some hidden fixed column width for the container?

## Expected Behavior

The full text should always be shown when the application window is wide enough.

v9.0.0 (commit `f25b8b1c`) amd64
Debian 11 Bullseye, KDE
[image]

iam-TJ, 2022-05-29:
Tracked the problem down to this part of code by increasing the fixed `150` to `300`:
```
$ git diff -- ./StandAlone/share/qml
diff --git a/StandAlone/share/qml/main.qml b/StandAlone/share/qml/main.qml
index 65a8c5b2..d164dbe2 100644
--- a/StandAlone/share/qml/main.qml
+++ b/StandAlone/share/qml/main.qml
@@ -140,7 +140,7 @@ ApplicationWindow {
     }
     ColumnLayout {
         id: leftControls
-        width: AkUnit.create(150 * AkTheme.controlScale, "dp").pixels
+        width: AkUnit.create(300 * AkTheme.controlScale, "dp").pixels
         anchors.top: parent.top
         anchors.topMargin: AkUnit.create(16 * AkTheme.controlScale, "dp").pixels
         anchors.left: parent.left
```
[image]

hipersayanX, 2022-05-29:
This is a known bug. The problem, I think, is that the Label is not reporting the text size properly, and that's why the ColumnLayout is smaller than it should be. But it does not happens with all operating systems or desktops environments, if I remember well, Mac and Windows worked fine, Arch with KDE also worked fine (I'm using it right now), Ubuntu had that bug too, in Fedora I don't remember, I did not pay attention to it.

iam-TJ, 2022-05-29:
Ubuntu would most likely inherit  any underlying Qt bug from Debian. I'll take a look later when I've solved the bigger issues - maybe some library needs a fix!

iam-TJ, 2022-05-29:
Unless I'm missing something, I believe I've fixed this using `childrenRect` to [size the column based on its children](https://doc.qt.io/qt-5/qml-qtquick-item.html#childrenRect-prop).
```
$ git show HEAD
commit 14aa2edad9fcef4e600587c5ce18523720d42ecf (HEAD -> issue529-truncated-GUI-use-flash)
Author: Tj <hacker@iam.tj>
Date:   Sun May 29 22:45:40 2022 +0100

    leftControls: size Column to not ellide Flash switch label

diff --git a/StandAlone/share/qml/main.qml b/StandAlone/share/qml/main.qml
index 65a8c5b2..344a0d0a 100644
--- a/StandAlone/share/qml/main.qml
+++ b/StandAlone/share/qml/main.qml
@@ -140,7 +140,7 @@ ApplicationWindow {
     }
     ColumnLayout {
         id: leftControls
-        width: AkUnit.create(150 * AkTheme.controlScale, "dp").pixels
+        width: AkUnit.create(childrenRect.width * AkTheme.controlScale, "dp").pixels
         anchors.top: parent.top
         anchors.topMargin: AkUnit.create(16 * AkTheme.controlScale, "dp").pixels
         anchors.left: parent.left
@@ -160,7 +160,6 @@ ApplicationWindow {
             id: chkFlash
             text: qsTr("Use flash")
             checked: true
-            Layout.fillWidth: true
             visible: false
             Accessible.name: text
             Accessible.description: qsTr("Use flash when taking a photo")
```
[image]

hipersayanX, 2022-05-30:
Just in case, doesn't that code fix throw some errors in terminal similar to:

```
Qt Quick Layouts: Polish loop detected. Aborting after two iterations.
```

iam-TJ, 2022-05-30:
No - nothing at all. Just tested with both a QtCreator Debug, and a manual cmake 'release' build. For avoidance of doubt I have the following three patches on top of master:
```
$ git status -u no; gitlog -n 8
On branch testing
nothi$ git status -u no; gitlog -n 8
On branch testing
nothing to commit, working tree clean
386339ec 2022-05-30 18:21:12 +0100 N Tj Merge branch 'issue529-truncated-GUI-use-flash' into testing
9946f4f1 2022-05-30 18:20:54 +0100 N Tj Merge branch 'issue528-fix-pixel-format-dialog' into testing
5576fbe4 2022-05-30 18:20:32 +0100 N Tj Merge branch 'issue530-add-cmake-target-uninstall' into testing
bbea5f60 2022-05-30 18:19:31 +0100 N Tj VideoOutputAddEdit: fix pixel format size display
e2aa0fa0 2022-05-30 15:28:26 +0100 N Tj leftControls: size Column to not ellide Flash switch label
a4c6a6fb 2022-05-29 22:08:15 +0100 N Tj cmake: add uninstall build target to aid development cycles
f25b8b1c 2022-05-19 07:19:12 +0200 E AHOHNMYC Translated using Weblate (Russian)
7c68a5b1 2022-04-27 14:14:46 +0200 E Yaron Shahrabani Translated using Weblate (Hebrew)ng to commit, working tree clean
386339ec 2022-05-30 18:21:12 +0100 N Tj Merge branch 'issue529-truncated-GUI-use-flash' into testing
9946f4f1 2022-05-30 18:20:54 +0100 N Tj Merge branch 'issue528-fix-pixel-format-dialog' into testing
5576fbe4 2022-05-30 18:20:32 +0100 N Tj Merge branch 'issue530-add-cmake-target-uninstall' into testing
bbea5f60 2022-05-30 18:19:31 +0100 N Tj VideoOutputAddEdit: fix pixel format size display
e2aa0fa0 2022-05-30 15:28:26 +0100 N Tj leftControls: size Column to not ellide Flash switch label
a4c6a6fb 2022-05-29 22:08:15 +0100 N Tj cmake: add uninstall build target to aid development cycles
f25b8b1c 2022-05-19 07:19:12 +0200 E AHOHNMYC Translated using Weblate (Russian)
7c68a5b1 2022-04-27 14:14:46 +0200 E Yaron Shahrabani Translated using Weblate (Hebrew)
```

hipersayanX, 2022-05-30:
Then do a PR.

iam-TJ, 2022-06-03:
Closed by issue #534

closed 2022-06-03 by iam-TJ