webcamoid/webcamoid #530: 9.0.0 - no Uninstall target for (c)make

iam-TJ, 2022-05-29:
## Summary

Whilst doing repeated build-install cycles during git-bisect operations the lack of an `uninstall` option for (c)make means having to do it manually. I'm currently doing:
```
$ find /usr/local/ -iname '*webcamoid*' -o -name '*avk*' -o -wholename '*avkys/*' -delete
```

## Expected Behavior

There should be an Uninstall option in the `Makefile`.

hipersayanX, 2022-05-29:
Will check, but that's not a supported feature for Cmake, [it looks kinda tricky](https://gitlab.kitware.com/cmake/community/-/wikis/FAQ#can-i-do-make-uninstall-with-cmake) to support it, and it seem prone to cause more bugs than solutions.

iam-TJ, 2022-05-29:
Indeed, CMake is a pain for this! There's a [StackOverflow article and response](https://stackoverflow.com/posts/69804923/revisions) with what looks like the most elegant potential partial solution that is added to `CMakeLists.txt` which I'll experiment with:
```
# make uninstall
add_custom_target("uninstall" COMMENT "Uninstall installed files")
add_custom_command(
    TARGET "uninstall"
    POST_BUILD
    COMMENT "Uninstall files with install_manifest.txt"
    COMMAND xargs rm -vf < install_manifest.txt || echo Nothing in
            install_manifest.txt to be uninstalled!
)```

iam-TJ, 2022-05-29:
So far this seems to be working well:
```
$ cmake --build ../builds/webcamoid --target uninstall
...
-- Build files have been written to: /srv/NAS/Sunny/SourceCode/builds/webcamoid
Uninstall files with install_manifest.txt
removed '/usr/local/lib/libavkys.so.9.0.0'
removed '/usr/local/lib/libavkys.so.9'
removed '/usr/local/lib/libavkys.so'
...
removed '/usr/local/bin/webcamoid'
...
removed '/usr/local/share/man/man1/webcamoid.1.gz'
removed '/usr/local/share/licenses/webcamoid/COPYING'
removed '/usr/local/share/applications/webcamoid.desktop'
Built target uninstall

```

closed 2023-06-10 by hipersayanX