webcamoid/webcamoid #531: 9.0.0 - Out-of-tree builds updating files in source repository

iam-TJ, 2022-05-29:
## Summary

Whilst doing repeated `git-bisect` operations and out-of-tree builds I noticed that three files are repeatedly written in the source repository but not in the out-of-tree build directory:
```
$ git status
HEAD detached at 2cb2f47f
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   StandAlone/share/ts/et.qm
        modified:   StandAlone/share/ts/he.qm
        modified:   StandAlone/share/ts/nl.qm
```
And the files:
```
$ md5sum ../builds/webcamoid/StandAlone/share/ts/{et,he,nl}.qm ./StandAlone/share/ts/{et,he,nl}.qm
md5sum: ../builds/webcamoid/StandAlone/share/ts/et.qm: No such file or directory
md5sum: ../builds/webcamoid/StandAlone/share/ts/he.qm: No such file or directory
md5sum: ../builds/webcamoid/StandAlone/share/ts/nl.qm: No such file or directory
3dfdcce83f10c50e3d8105b8f9128836  ./StandAlone/share/ts/et.qm
aafd420e5d5fe518cfcabb2863f7b680  ./StandAlone/share/ts/he.qm
d5eb6b453eadc6ab78aa42f0edc3b8df  ./StandAlone/share/ts/nl.qm
```
Build command is:
```
$ ( B="../builds/webcamoid"; git status; git show -s --oneline $(git status | head -1 | cut -d " " -f 4); echo "INFO: Configuring in $B"; cmake -S . -B $B; sleep 5; echo "INFO: Building to $B"; cmake --build $B --parallel 12; sleep 5; echo "INFO: Installing from $B"; cmake --install $B; ) | tee /tmp/webcamoid-build.log
```
Build products are under `../builds/webcamoid/`.

Using Debian 11 Bullseye on amd64.

hipersayanX, 2022-05-29:
This a known error by design. When compiling, it takes the .ts files and compile to .qm, because I want them to be embedded in Webcamoid's binary file, the resulting files should be put relative to the [.qrc file](https://github.com/webcamoid/webcamoid/blob/master/StandAlone/translations.qrc), that's why the source tree is modified.
The right way is not embedding the files, but instead installing the files on the install directory.

iam-TJ, 2022-06-10:
I've been doing some follow-up on this and, as far as I can tell, a one-line change in `StandAlone/Translations/CMakeLists.txt` is all that is required (on our side!).
```
$ git diff StandAlone/Translations/CMakeLists.txt
diff --git a/StandAlone/Translations/CMakeLists.txt b/StandAlone/Translations/CMakeLists.txt
index 1a6219ca..7260659b 100644
--- a/StandAlone/Translations/CMakeLists.txt
+++ b/StandAlone/Translations/CMakeLists.txt
@@ -80,7 +80,7 @@ if (LRELEASE_TOOL)
         add_custom_target(tranlation_${LANG}_qm ALL
                           ${LRELEASE_TOOL} -removeidentical
                                            -compress ${CMAKE_SOURCE_DIR}/StandAlone/share/ts/${LANG}.ts
-                                           -qm ${CMAKE_SOURCE_DIR}/StandAlone/share/ts/${LANG}.qm
+                                           -qm ${CMAKE_BINARY_DIR}/StandAlone/share/ts/${LANG}.qm
                           DEPENDS tranlation_${LANG}_ts
                           BYPRODUCTS StandAlone/share/ts/${LANG}.qm
                           VERBATIM)
```

After out-of-tree builds I've confirmed that the `.qm` binaries are only written into the build path (I delete any `*.qm` files from the source tree before the build):
```
SourceCode/webcamoid$ find . ../builds/webcamoid/ -name '*.qm'
../builds/webcamoid/StandAlone/share/ts/si.qm
../builds/webcamoid/StandAlone/share/ts/hu.qm
../builds/webcamoid/StandAlone/share/ts/nb_NO.qm
../builds/webcamoid/StandAlone/share/ts/pt_BR.qm
...
```
Unfortunately [there's a bug in the `cmake` autogenerator code I've now reported](https://gitlab.kitware.com/cmake/cmake/-/issues/23611) that, even with out-of-tree builds that put the `*.qm` files in the correct build location, generates an RCC `*_Info.json` containing absolute source directory paths.

However, working around that by putting the `*.qm` binary translation files in the source tree so the executable builds and links successfully, when the binary is executed from the build path with e.g:
```
LANGUAGE=de_DE.UTF ../builds/webcamoid/build/bin/webcamoid
```
it correctly switches to German language without loading any external files - confirmed by executing it under `strace` and checking what files it opens:
```
LANGUAGE=nl_NL.UTF-8 strace -f -e trace=open,openat  ../builds/webcamoid/build/bin/webcamoid  |& tee /tmp/strace.log
```
I've tested other languages successfully so I'm working on a simple `cmake` action to post-process the `*_Info.json` file to correct the path(s) to the `*.qm` files during the build.

hipersayanX, 2022-06-11:
Yes, that's just one of the fixes that must be done, but then you must remove the qrc file, install the qm files into /usr/share/tr (or whatever, also depending on the platform), and Webcamoid must find the translation files relative to the binary for portable versions to work properly.

iam-TJ, 2022-06-12:
I've now got out-of-tree builds working correctly with `.qrc` files being generated at `configure` time and translation `.qm` files written into the build tree, not the source tree. This is still including the binary translation files directly in the libraries/executable.

I've also submitted a patch to `cmake` as [merge request 7352 "Qt: AUTORCC: allow relative paths in .qrc files"](https://gitlab.kitware.com/cmake/cmake/-/merge_requests/7352) that adds  `AUTORCC_CANONICAL_PATHS` variable which defaults to `ON` and must be set `OFF` to retain relative paths which helps make out-of-tree builds that use AUTORCC easier.

With a few changes to some `CMakeLists.txt` and replacing a handful of `.qrc` files with `.qrc.in` which include `\${CMAKE_CURRENT_BINARY_DIR}` path prefixes which are processed using `configure_file()` and written to the build tree, the project builds whether or not `AUTORCC_CANONICAL_PATHS` is set, and is backward compatible with existing CMake versions.

I'm going to do some more testing with permutations of build options to ensure there are no regressions, and eventually I'll submit a PR for this work (which won't rely on CMake having `AUTORCC_CANONICAL_PATHS`).

closed 2023-06-10 by hipersayanX