webcamoid/webcamoid #545: Plugins initialising properties twice

iam-TJ, 2022-06-06:
Whilst developing a new **Crop** plugin and debugging some complex interactions between weakly related properties between C++ and QML, I noticed that when the plugin is being initialised the `updateEffectsProperties()` is being called twice, resulting in the plugin's `set*()` properties called twice.

I'm attaching two screenshots of QtCreator's Debug mode. Note the repeated calls to the `set<property>()` methods.
```
debug: Crop::setHeight( 1080 )
debug: Crop::setLeft( 0 )
debug: Crop::setTop( 0 )
debug: Crop::setWidth( 1920 )
*** first screenshot captured ***
debug: Crop::setHeight( 1080 )
debug: Crop::setLeft( 0 )
debug: Crop::setTop( 0 )
debug: Crop::setWidth( 1920 )
*** second screenshot captured ***
```
The stack traces for the two screenshots show `VideoEfffectsPrivate::updateEffectsProperties()` is called twice. With breakpoint at `CropElement::setWidth()` the first stack trace shows:
```
CropElement::setWidth
CropElement::qt_static_metacall
CropElement::qt_metacall
QMetaProperty::write
QObject::setProperty
VideoEffectsPrivate::updateEffectsProperties
VideoEffects::setEffects
VideoEffectsPrivate::updateEffects
VideoEffects::VideoEffects
MediaTools::init
main
```
[`VideoEffects::setEffects() calls updateEffectsProperties()`](https://github.com/webcamoid/webcamoid/blob/d885fd36ef662786ae9c5a28aa5df1a58f897df9/StandAlone/src/videoeffects.cpp#L265) via [`VideoEffects::VideoEffects()         this->d->updateEffects();;`](https://github.com/webcamoid/webcamoid/blob/d885fd36ef662786ae9c5a28aa5df1a58f897df9/StandAlone/src/videoeffects.cpp#L89).

The second breakpoint shows:
```
CropElement::setWidth
CropElement::qt_static_metacall
CropElement::qt_metacall
QMetaProperty::write
QObject::setProperty
VideoEffectsPrivate::updateEffectsProperties
VideoEffects::VideoEffects
MediaTools::init
main
```
Here the call to `updateEffectsProperties()` is repeated in the constructor at [`this->d->updateEffectsProperties();`](https://github.com/webcamoid/webcamoid/blob/d885fd36ef662786ae9c5a28aa5df1a58f897df9/StandAlone/src/videoeffects.cpp#L89) - the following line.
```
VideoEffects::VideoEffects(QQmlApplicationEngine *engine, QObject *parent):
    QObject(parent)
{
    this->d = new VideoEffectsPrivate(this);
...    
    this->d->updateEffects();
    this->d->updateEffectsProperties();
}
```

hipersayanX, 2022-06-11:
Ok, get it

```
VideoEffects::VideoEffects()
{
    ...

    this->d->updateEffects();
        // -> self->setEffects(effects);
            // -> this->d->updateEffectsProperties();

    /* Since this is called in the previous call, it should be removed.
     * You must check if it can load the previous effects settings when
     * loading Webcamoid.
     */ 
    this->d->updateEffectsProperties();
}
```

iam-TJ, 2022-06-12:
>```
>      * You must check if it can load the previous effects settings when
>      * loading Webcamoid.
> ```
Does this mean if I remove the 2nd call there needs to be some additional code to do this check?

hipersayanX, 2022-06-18:
Nope, just test manually if it keep working fine without breaking anything, if so do a PR.

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX