webcamoid/webcamoid #549: Using the portable version on Windows leaves a virtual webcam behind when finished.

timboxyz, 2022-06-13:
Using the Portable version on Windows, when finished the virtual camera remains behind appearing in lists of webcams with the default "no output" blue Webcamoid display. this even survives a reboot.

I would expect the portable version to leave nothing behind once the program terminates.

Easily reproduced by downloading and running the portable version, then quitting and looking at the list of available webcams in some other programme, e.g. OBS studio/Zoom etc

If the portable version installs some drivers for the virtual webcam then there should be some way to uninstall them.

* Webcamoid portable X64 8.8
* Windows 10 x64

hipersayanX, 2022-06-18:
Just uninstall [AkVirtualCamera](https://support.microsoft.com/en-us/windows/uninstall-or-remove-apps-and-programs-in-windows-10-4b55f974-2cc6-2d2b-d092-5905080eaf98).

timboxyz, 2022-06-19:
> Just uninstall [AkVirtualCamera](https://support.microsoft.com/en-us/windows/uninstall-or-remove-apps-and-programs-in-windows-10-4b55f974-2cc6-2d2b-d092-5905080eaf98).

Thanks for the suggestion. However, I would not expect a portable version to perform any install and there is no "AkVirtualCamera" showing in any of the 3 methods described in the provided link. Searching in the taskbar provides only the .exe located in the portable folder. There are multitudinous Registry entries for "Webcamoid".

I wonder therefore if upon exiting the portable version it tries to perform some sort of uninstall following a temporary install but fails to execute it correctly/completely.

hipersayanX, 2022-06-20:
Sorry, I did not read you were using an old version of Webcamoid, you can uninstall the vcam following the instructions in  #178. The bug you are reporting was already fixed with Webcamoid 9.

>I wonder therefore if upon exiting the portable version it tries to perform some sort of uninstall following a temporary install but fails to execute it correctly/completely.

The portable version can't perform any sort of uninstall step because, for "uninstalling", you just delete the files manually, in contrast with the vcam that must be installed and uninstalled properly. Again this was already fixed with Webcamoid 9, splitting the vcam into it's own project and adding it's installer/uninstaller, while still keeping the portable version on Webcamoid.

timboxyz, 2022-06-21:
Unfortunately, upgrading doesn't solve that sort of bug :(
Fortunately using [GraphStudioNext](https://github.com/cplussharp/graph-studio-next) as suggested in #178 seems to have solved the problem.

Thanks for your help.

closed 2022-06-23 by timboxyz