webcamoid/webcamoid #552: Webcamoid AppImage not opening/running properly on Fedora 36

r2rX, 2022-06-19:
Hello there,

After searching for a good webcam app, Webcamoid seems to fit the bill for my usecase. :) However, when opening [webcamoid-portable-linux-9.0.0-x86_64.AppImage](https://github.com/webcamoid/webcamoid/releases/download/9.0.0/webcamoid-portable-linux-9.0.0-x86_64.AppImage), the lights on the camera turn on but the UI doesn't load/display. The only recourse at this point is to kill the webcamoid process manually.

My system specs are:
-AMD Ryzen 5900HX & RX 6800M
-Logitech C930e Webcam
-Fedora 36 KDE (running Wayland)
  -Kernel 5.18.5-200
  -KDE Plasma v5.25.0
  -KDE Framework v5.95.0
  -Qt version 5.15.3
  -Wayland 1.20.0
  -Pipewire v0.3.52

Here's the terminal log:
```
[xxxxx@home Applications]$ ./webcamoid-portable-linux-9.0.0-x86_64_385e22076e260b0d88b643e34d78fdb8.AppImage 
/usr/bin/AppImageLauncher: /lib64/libcurl.so.4: no version information available (required by /usr/bin/../lib/x86_64-linux-gnu/appimagelauncher/libappimageupdate.so)
QSocketNotifier: Can only be used with threads started with QThread
[2022-06-19 14:53:25.255, webcamoid, 0x7fd2b5bcc7c0,  (0)] warning: QSocketNotifier: Can only be used with threads started with QThread
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libaccess_alsa_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libaccess_concat_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libaccess_imem_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libaccess_mms_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libaccess_mtp_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libaccess_realrtsp_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libattachment_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libavio_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libcdda_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libdc1394_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libdtv_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libdv1394_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libdvb_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libdvdnav_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libdvdread_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libfilesystem_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libftp_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libhttp_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libhttps_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libidummy_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libimem_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/liblibbluray_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/liblinsys_hdsdi_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/liblinsys_sdi_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/liblive555_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libnfs_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libpulsesrc_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/librtp_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libsatip_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libsdp_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libsftp_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libshm_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libtcp_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libtimecode_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libudp_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libv4l2_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libvcd_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access/libvdr_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access_output/libaccess_output_dummy_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access_output/libaccess_output_file_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access_output/libaccess_output_http_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access_output/libaccess_output_livehttp_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access_output/libaccess_output_shout_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access_output/libaccess_output_srt_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/access_output/libaccess_output_udp_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/audio_filter/libaudio_format_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/audio_filter/libaudiobargraph_a_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/audio_filter/libchorus_flanger_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/audio_filter/libcompressor_plugin.so
[... 292 lines cut ...]
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/video_filter/libscale_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/video_filter/libscene_plugin.so
[000055b013e4b3a0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcamZVlTw1/lib/vlc/plugins/video_filter/libsepia_plugin.so
[... cut ...]
[2022-06-19 14:53:30.256, Webcamoid, 0x7fd272ffd640,  (0)] debug: Camera input frame format: AkCaps(mimeType="video/unknown",fourcc=QVariant(QString, "YUYV"),fps=QVariant(QString, "30/1"),height=QVariant(uint, 480),width=QVariant(uint, 640))
```

AppImageLauncher is utilized for system intergration but the error is the same (minus the AppImageLauncher libcurl error) when running the AppImage directly (as well as via the portable folder download).

Please let me know if there's any further information I can provide that may help. Cheers!

bearsh, 2022-12-16:
same on kde neon which is based on ubuntu 20.04

hipersayanX, 2023-05-09:
The daily build is working fine with Fedora 38.

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX

OlliAtWork, 2023-06-12:
Seems like on KDE neon based on 22.04 this issue is still present. The Program starts with no error infos but no UI is presented. 
```
./Applications/webcamoid-portable-linux-9.1.0-x64_8df145ac5ffd4f71c51ab45330b3d9ea.AppImage 
[2023-06-12 08:28:26.164, Webcamoid, 0x7faf667507c0,  (0)] debug: Multiplanar format reported as single planar: nv12
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libaccess_alsa_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libaccess_concat_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libaccess_imem_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libaccess_mms_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libaccess_mtp_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libaccess_realrtsp_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libattachment_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libavio_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libcdda_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libdc1394_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libdtv_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libdv1394_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libdvb_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libdvdnav_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libdvdread_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libfilesystem_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libftp_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libhttp_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libhttps_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libidummy_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libimem_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/liblibbluray_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/liblinsys_hdsdi_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/liblinsys_sdi_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libnfs_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libpulsesrc_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/librist_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/librtp_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libsatip_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libsdp_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libsftp_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libshm_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libtcp_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libtimecode_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libudp_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libv4l2_plugin.so
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/access/libvcd_plugin.so
[... cut ...]
[00005560b6503cd0] main libvlc error: stale plugins cache: modified /tmp/.mount_webcam94Rf9V/lib/vlc/plugins/video_output/libyuv_plugin.so
[2023-06-12 08:28:27.901, Webcamoid, 0x7faf667507c0,  (0)] warning: QQmlApplicationEngine failed to load component
[2023-06-12 08:28:27.901, Webcamoid, 0x7faf667507c0, main.qml (590)] warning: qrc:/Webcamoid/share/qml/main.qml:590:5: Type RunCommandDialog unavailable
[2023-06-12 08:28:27.901, Webcamoid, 0x7faf667507c0, RunCommandDialog.qml (23)] warning: qrc:/Webcamoid/share/qml/RunCommandDialog.qml:23:1: module "QtQuick.Controls.Styles" is not installed

hipersayanX, 2023-06-12:
@OlliAtWork is because I forgot to remove [this line](https://github.com/webcamoid/webcamoid/blob/3f2228b8e61c4cf636f040c2a10a52bcbb56ddf4/StandAlone/share/qml/RunCommandDialog.qml#L23). But I will take the chance and also try to release the ARM packages that I wasn't able to release in this one, so it will take a few days.

OlliAtWork, 2023-06-26:
I can confirm that now it works as expected. Thanks!