webcamoid/webcamoid #553: Webcamoid crashes indefinitely after selecting the AV1 codec

branches78, 2022-06-23:
## Summary

Webcamoid crashes indefinitely after selecting the AV1 codec, the app won't open anymore and a gnome ABRT notification appears stating that the app crashed but a bug report can't be sent from there.

## Current Behavior

The app won't open anymore after I selected the AV1 codec in the "Video Recording" tab.

## Expected Behavior

Continue using the app to take videos using the AV1 codec.

How it should have work?

After I selected the AV1 codec and click on "close" I should have been able to continue using the app.

## Steps to Reproduce

1.  Click on the cog icon at the top right corner of the app
2.  Click on preferences
3.  Click on "Video Recording" tab on the left
4.  On the current file format that appears I just clicked on the drop-down menu called "video codec" and selected AV1, the rest was untouched.
5. Click on "close" at the bottom right of the window.

## Your Environment

* Webcamoid information: version 9.0.0 for x64
* Operating System information: Fedora 36 Workstation Edition
* Any other useful information: my computer is an 11th Gen Intel® Core™ i3-1115G4 × 4 with the Mesa Intel® UHD Graphics (TGL GT2).

yatian-liu, 2023-01-30:
Can confirm this problem on Arch Linux. If the libaom AV1 codec is selected, the app will crash immediately after clicking "close"; if the svt-av1 codec is selected, the app will not crash after clicking "close", but will crash after clicking the "start recording" button for video. Removing the config file and start fresh will prevent the app from crashing, which further confirms the problem is the AV1 codec support. Both libaom-av1 and libsvt-av1 are installed.

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX