webcamoid/webcamoid #554: Does not see USB microphone

holmesrichards, 2022-07-05:
## Summary

Webcamoid does not recognize USB microphone as a sound source

## Current Behavior

Under Audio > Sources, only settings are Silence, Monitor of Built-in Analog Stereo, Built-in Analog Stereo, and HD Pro Webcam C920 Analog Stereo

## Expected Behavior

I would expect it to also include my TONOR TC30 USB microphone.

## Your Environment

* Webcamoid information: Version 9.0.0
* Operating System information: Linux Mint 20.3 Cinnamon 
* Any other useful information: System sound settings showing USB microphone, Webcamoid audio settings not showing it
[image]

hipersayanX, 2022-07-18:
What is the output of these commands?

```
pacmd list-cards
pacmd list-sources
```

holmesrichards, 2022-07-18:
The mic is now showing in webcamoid. I don't know what's changed.

closed 2022-07-18 by holmesrichards