webcamoid/webcamoid #558: Not able to drag window to another monitor, not able to fullfill a half monitor size

duytrung, 2022-07-27:
## Summary

I have to monitors and I could not drag the app windows to another monitor. 

## Current Behavior

Dragging to another monitor will fling the windows back to its usual position
## Expected Behavior

How it should have work?

## Steps to Reproduce

Click and drag the app windows to another monitor.

## Suggestions and tips

## Your Environment

* Webcamoid information: (version) 9 - windows 11.
* Operating System information: Windows 11
* Any other useful information: 

Thanks for considering the issue

RouHim, 2022-08-05:
Same happens on Linux.

digitalextremist, 2023-02-23:
Seeing the same with `Mint 20.3`

francoisbillard, 2023-03-12:
Seeing same issue with Mint 21.1. unusable with two screens ...

digitalextremist, 2023-03-12:
Never found the solution to this, but more so had serious resource issues and horrible quality with audio and video losing sync.

My reaction was to just use `OBS` and all is solved, with far better functionality and quality

https://github.com/obsproject/obs-studio