webcamoid/webcamoid #559: 9.0.0 - FTBFS with ffmpeg-5.1

grover92000, 2022-08-04:
Build fails with ffmpeg-5.1:

```
[ 18%] Building CXX object libAvKys/Plugins/MultiSrc/src/ffmpeg/CMakeFiles/Multi
Src_ffmpeg.dir/src/mediasourceffmpeg.cpp.o
In file included from /tmp/build/webcamoid-9.0.0/libAvKys/Plugins/MultiSrc/src/ffmpeg/src/audiostream.h:23,
                 from /tmp/build/webcamoid-9.0.0/libAvKys/Plugins/MultiSrc/src/ffmpeg/src/mediasourceffmpeg.cpp:40:
/tmp/build/webcamoid-9.0.0/libAvKys/Plugins/MultiSrc/src/ffmpeg/src/abstractstream.h:60:21: error: ‘AVCodecContext’ does not name a type; did you mean ‘AVIODirContext’?
   60 |         Q_INVOKABLE AVCodecContext *codecContext() const;
      |                     ^~~~~~~~~~~~~~
      |                     AVIODirContext
/tmp/build/webcamoid-9.0.0/libAvKys/Plugins/MultiSrc/src/ffmpeg/src/abstractstream.h:71:42: error: ‘AVSubtitle’ has not been declared
   71 |         Q_INVOKABLE void subtitleEnqueue(AVSubtitle *subtitle);
      |                                          ^~~~~~~~~~
/tmp/build/webcamoid-9.0.0/libAvKys/Plugins/MultiSrc/src/ffmpeg/src/abstractstream.h:84:34: error: ‘AVSubtitle’ has not been declared
   84 |         virtual void processData(AVSubtitle *subtitle);
      |                                  ^~~~~~~~~~
In file included from /tmp/build/webcamoid-9.0.0/libAvKys/Plugins/MultiSrc/src/ffmpeg/src/mediasourceffmpeg.cpp:42:
/tmp/build/webcamoid-9.0.0/libAvKys/Plugins/MultiSrc/src/ffmpeg/src/subtitlestream.h:43:26: error: ‘AVSubtitle’ has not been declared
   43 |         void processData(AVSubtitle *subtitle);
      |                          ^~~~~~~~~~
make[2]: *** [libAvKys/Plugins/MultiSrc/src/ffmpeg/CMakeFiles/MultiSrc_ffmpeg.dir/build.make:146: libAvKys/Plugins/MultiSrc/src/ffmpeg/CMakeFiles/MultiSrc_ffmpeg.dir/src/mediasourceffmpeg.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3572: libAvKys/Plugins/MultiSrc/src/ffmpeg/CMakeFiles/MultiSrc_ffmpeg.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

grover92000, 2022-08-12:
I've fixed the problem and attached my patch.

[webcamoid-9.0.0-ffmpeg-5.1.patch.txt](https://github.com/webcamoid/webcamoid/files/9329193/webcamoid-9.0.0-ffmpeg-5.1.patch.txt)

hipersayanX, 2022-08-13:
Can you make a PR, please?

grover92000, 2022-08-13:
Done  @ https://github.com/webcamoid/webcamoid/pull/560

hipersayanX, 2022-08-14:
Merged, thanks!

closed 2022-08-14 by hipersayanX