webcamoid/webcamoid #565: Many errors [main libvlc error: stale plugins cache]  ...but works fine.

AndKe, 2022-09-28:
## Summary

When running 
./webcamoid-portable-linux-9.0.0-x86_64.AppImage 

I see lots of 
```
[0000559eafe03b40] main libvlc error: stale plugins cache: modified /tmp/.mount_webcammn4wqc/lib/vlc/plugins/video_filter/librotate_plugin.so
[0000559eafe03b40] main libvlc error: stale plugins cache: modified /tmp/.mount_webcammn4wqc/lib/vlc/plugins/video_filter/libscale_plugin.so

```
(many, many more, with different files.)

Also: it seems to work fine.  :) 

## Your Environment

Ubuntu 22.04

hipersayanX, 2022-09-28:
Don't worry, these are just warnings related to VLC cache, these messages always appears in the AppImage because the mount folder changes every time you launch Webcamoid, isn't possible to fix or hide the messages, but the program works fine.

AndKe, 2022-09-28:
ok, thank you.

closed 2022-09-28 by AndKe