webcamoid/webcamoid #566: Windows installer does not accept arguments any more

riedel, 2022-10-14:
## Summary
The version 8 installer supported passing arguments such as --script for silent install to the installer on Windows

## Current Behavior

The current installer executables seem to ignore any arguments passed via the command line

## Expected Behavior

QT Installer framework arguments should work: https://doc.qt.io/qtinstallerframework/ifw-cli.html

## Steps to Reproduce
`webcamoid-installer-windows-9.0.0-win32.exe --verbose --script silentInstall.js`

## Your Environment

* Webcamoid information: 9.0.0
* Operating System information: Windows 10

hipersayanX, 2022-10-15:
Webcamoid 8 used [QtIFW](https://download.qt.io/official_releases/qt-installer-framework/), meanwhile Webcamoid 9 uses [NSIS](https://nsis.sourceforge.io/Download), [here](https://nsis.sourceforge.io/Docs/Chapter3.html) is the documentation for the installer options.

riedel, 2022-10-15:
Thanks a lot! This explains it.

closed 2022-10-15 by riedel