webcamoid/webcamoid #567: The virtual camera is not installed, although both v4l2loopback and akvcam are loaded.

MordragT, 2022-11-01:
## Summary

The virtual camera is not installed, although both v4l2loopback and akvcam are loaded.

## Current Behavior

In the general settings both v4l2looppback and akvcam are shown aswell as pkexec as root method.
When reloading the akvcam module both virtual cameras (see config below) are correctly created.

## Expected Behavior

Be able to create video outputs from webcamoid

## Your Environment

I tested it with both the master revision and version 9.

OS: NixOS 22.11 unstable

This is how my derivation in nix looks like:

```nix
  pname = "webcamoid";
  version = "master";

  src = fetchFromGitHub {
    sha256 = "JmItbJmEMmDirIaOV4XKF43/PrCGwHB4MHB+SlGMy8I=";
    rev = version;
    repo = "webcamoid";
    owner = "webcamoid";
  };

  buildInputs = [
    libxcb
    qtbase
    qtdeclarative
    qtquickcontrols
    qtquickcontrols2
    ffmpeg-full
    gst_all_1.gstreamer
    gst_all_1.gst-plugins-base
    alsa-lib
    libpulseaudio
    jack2
    v4l-utils
    kmod.dev
  ];

  nativeBuildInputs = [ pkg-config cmake wrapQtAppsHook ];

  #dontWrapQtApps = true;

  cmakeFlags = [
    "-DEXTRA_SUDOER_TOOL_DIR=${polkit.bin}/bin/" # directory contains pkexec
  ];
```

And this is where I load both akvcam and v4l2loopback:

```nix
  boot.extraModulePackages = with config.boot.kernelPackages; [ akvcam v4l2loopback ];
  boot.kernelModules = [ "akvcam" "v4l2loopback" ]; # this adds both modules to /etc/modules-load.d/nixos.conf
  environment.systemPackages = with pkgs; [
    webcamoid # installs the former defined package
  ];

  environment.etc."akvcam/config.ini" = {
    text = "[Cameras]
cameras/size = 2

cameras/1/type = output
cameras/1/mode = mmap, userptr, rw
cameras/1/description = Akvcam (Output device)
cameras/1/formats = 1

cameras/2/type = capture
cameras/2/mode = mmap, rw
cameras/2/description = Akvcam (Capture device)
cameras/2/formats = 1

[Formats]
formats/size = 1

formats/1/format = YUY2
formats/1/width = 1280
formats/1/height = 720
formats/1/fps = 30

[Connections]
connections/size = 1
connections/1/connection = 1:2";
    mode = "0644";
  };
```

I know that my environment might be unusal and I tried my best to find the issue on my own, but maybe I am missing something obvious :)

hipersayanX, 2022-11-02:
Clone of #510, read [this](https://github.com/webcamoid/webcamoid/issues/510#issuecomment-1107249765), closing.

closed 2022-11-02 by hipersayanX

Tomaszal, 2024-11-23:
This is still an issue on NixOS despite #510 being closed. I'm also not sure that was the same issue to begin with. I have both `v4l2loopback` and `akvcam` installed, loaded, and the virtual cameras show up with `v4l2-ctl --list-devices` as expected:

```
Virtual Camera (platform:akvcam-3):
	/dev/video3

Virtual Camera (output device) (platform:akvcam-7):
	/dev/video7

Dummy video device (0x0000) (platform:v4l2loopback-000):
	/dev/video0
```

Despite that, when picking video outputs, Webcamoid says `The virtual camera is not installed. Please install v4l2loopback` or  `The virtual camera is not installed, do you want to install it?` if `akvcam` driver is selected. Installing it via that prompt fails, but that's sort of expected on NixOS and is a separate issue