webcamoid/webcamoid #568: Memory leak? Windows version freezes upon start, builds up RAM usage until 100% avail. RAM

lumihei, 2022-11-02:
# Reporting a problem?

## Summary
When I start version 9.0 under Win 10, it takes a while until the windows is visible, then freezes upon showing the main window. Task Manager shows climbing amount of RAM use, until it maxes out with all avail. RAM of my machine (32 GB).

Tried daily build, same behaviour. Pity as I can't get it working.

## Steps to Reproduce
Start the program

## Your Environment
* Webcamoid information: (version) (architecture) - see above
* Operating System information: Win 10, Version 22H2 Build 19045.2130

lumihei, 2022-11-02:
PS.: Win 64bit Version

jabbermacy, 2023-02-24:
same issue here, Win 11

hipersayanX, 2023-03-05:
For people with this problem:

>Did you did something something special for this to happen?
>You installed Webcamoid for the first time and this happened as-is without any kind of intervention?
>Maybe this happened after installing the virtual camera?
>Did you tested the daily build?

I need a lot more information because all the information provided until now y very vague and can't be reproduced. Therefore, no information, no fix.

fohlsso2, 2023-03-17:
Did some testing yesterday as I came across this software hoping it would solve an issue i have with my current solution.
With release version it stops working when the virtual camera is installed, uninstall camera and it no longer hogs all ram.
Also tried 32bit version and insted of hogging all ram it crashes (although I did not check wich camera binary it tried to load)
All config as default.
Daily build of camera- no change.
Daily build of webcamoid does not seem to suffer from this particular issue.

After initial install and subsequent manual uninstall of virtual camera it is no longer possible to install it again through the gui. It downloads the installer, and the admin prompt is shown but then it just stays there and the installer never actually starts doing anything (it is visible in taskmanager so it is loaded).

And unfortunately I will have to leave you there, because even when using the portable version it messes with other QT based applications and breaks them.

But I hope at least you have a little more information. All above tested on win10 enterprise, build 19044.2604. HP ZBook i7 laptop.

hipersayanX, 2023-03-19:
>And unfortunately I will have to leave you there, because even when using the portable version it messes with other QT based applications and breaks them.

In which way? did the apps stopped working, crashed? which apps did you tested?

fohlsso2, 2023-03-19:
its a couple work related programs, a cad software and a simlutation package.  a simple repair/re-install solved both. But neither of them ran, they both crashed when trying to start. 
If it was webcamoid or virtual camera that caused it I don't know. ( since virtual camera is the one with an installer my vote would be on it, but  I havent verified it so can't tell for sure).

hipersayanX, 2023-03-19:
>If it was webcamoid or virtual camera that caused it I don't know. ( since virtual camera is the one with an installer my vote would be on it, but I havent verified it so can't tell for sure).

If the client program crashed, it was the virtual camera, but can't help much if I don't know which program it was.

geronimo8, 2023-04-27:
I had the same issue as well. 
It's the akvirtualcamera-windows-9.1.0.exe. I tried using webcamoid 9.0.0 (installer and portable) by itself, and it ran fine. I installed akvirtualcamera, then webcamoid would just hang on startup, while consuming all the remaining RAM. I uninstalled akvirtualcamera and deleted the folder, webcamoid ran fine again.
Win 10, Version 22H2 Build 19045.2846

RZMiRaN, 2023-05-24:
Same here, only happens when akvirtualcamera is installed.

hipersayanX, 2023-06-10:
Should be fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX