webcamoid/webcamoid #569: can't select webcamera

dchmelik, 2022-11-09:
## Summary

Can't select webcamera.

## Current Behavior

Chooses built-in webcamera and doesn't let you select another (USB, etc., unless hidden in extremely difficult/slow-to-use/-find  new-/alternative-style menus copied from Apple's & Google's number one top worst design style).

## Expected Behavior

Allow selecting webcamera.

## Your Environment

* Webcamoid information: 9
* Operating System information: Windows 10 x86(-64) (willing to try on UNIX/GNU/Linux if a FreeBSDPackage or Slack(Build|Package))
* Lenovo X1 Yoga ThinkPad, Logitech QuickCam Pro 9000

# Want a new feature?

I guess new but, very basic!

hipersayanX, 2023-05-11:
I can't reproduce this issue, all cameras can be detected fine and can be selected, and working.

[image]

Maybe I'm missing context here?

dchmelik, 2023-05-12:
I see.  Since I was taking photos, not making videos, that's unclear.  Classic (File Edit View Help, etc.) menus are best and clearer.

closed 2023-05-12 by dchmelik

hipersayanX, 2023-05-12:
>Classic (File Edit View Help, etc.) menus are best and clearer.

The problem is that traditional menus doesn't fit well in small screens, the best option is using context menus and choosing carefully the design for not overloading the UI, and still having the most important options quickly accessible.

[image]