webcamoid/webcamoid #570: no photographing

dchmelik, 2022-11-09:
## Summary

No longer photographing.

## Current Behavior

Press 'take photo' multiple times: nothing happens.  There's no confirmation/thumbnail; no file appears in user's home directory/folder (nor Pictures folder/directory, which shouldn't need exist to save) when its remaining free storage capacity is several GB.
## Expected Behavior

Photograph.

## Your Environment

* Webcamoid information: 9
* Operating System information: Windows 10 x86(-64) (willing to try on UNIX/GNU/Linux if a FreeBSDPackage or Slack(Build|Package))
* Lenovo X1 Yoga ThinkPad, Logitech QuickCam Pro 9000

dchmelik, 2022-11-12:
> Try the following, go to the _menu > Preferences > Image capture > Images directory > Search_ and select the Images directory.

That should be entirely unnecessary--as in webcamoid v8.8--so I'll stick with that for now.

hipersayanX, 2022-11-13:
I did tried and, beside the frame saving animation isn't running, the frame is being properly saved in the Pictures folder. So it's working for me (except for the animation).

>That should be entirely unnecessary

Yes, it's necessary to check if you have the saving folder properly configured and discard a bug in my side.

>--as in webcamoid v8.8--so I'll stick with that for now.

Ok, but that version won't receive updates anymore and bugs reports will be totally ignored.

Anutrix, 2022-12-27:
I'm new to Webcamoid but if there used to be capture animation, it doesn't work now. At least on 9.0 on Win11.

hipersayanX, 2022-12-28:
Could this issue be related to #577?

Anutrix, 2022-12-29:
> Could this issue be related to #577?

Partially related maybe. The no capture animation issue part is unrelated to it. But the part where image is not created/stored at all is probably same.

We need a log section of some kind inside app itself that advanced users can choose to enable and see.

hipersayanX, 2023-01-04:
@Anutrix 

>We need a log section of some kind inside app itself that advanced users can choose to enable and see.

I know, that's a pending feature.

commit fd271743 (2023-05-11): Do a proper conversion between local file path and url format (fixes #570 and #577).
closed 2023-05-11 by hipersayanX