webcamoid/webcamoid #571: allow standard window position/movement/resize

dchmelik, 2022-11-09:
# Summary

On non-Apple, doesn't respect when system has taskbar in Apple position (top) nor allows most resultantly necessary moving/resizing/maximizing/minimizing.

## Current Behaviour

On non-Apple, doesn't respect when system has taskbar in Apple position (top) nor allows most resultantly necessary moving/resizing/maximizing/minimizing.

## Expected Behaviour

Check taskbar position and position accordingly, and allow moving/resizing/maximizing/minimizing.

## Steps to Reproduce

* Webcamoid information: 8, 9
* Operating System information: Windows 10 x86(-64) (willing to try on UNIX/GNU/Linux if a FreeBSDPackage or Slack(Build|Package))
* Lenovo X1 Yoga ThinkPad, Logitech QuickCam Pro 9000

# Want a new feature?

I guess new, but most basic!

hipersayanX, 2022-11-09:
I don't understand this, the minimize, maximize, close buttons does not show in the window? the window draw is controlled by the window manager, I haven't any control of that in the code. Can you show me a screen shot?

dchmelik, 2022-11-11:
The buttons show, but don't work, like they do in most other programs, but if it's true only the window manager controls it then (as long as you test in Windows) maybe this can be closed.

hipersayanX, 2022-11-13:
>The buttons show, but don't work, unlike in most other programs

It works exactly the same way, I've tried and did not found any bug. Could it be may that Webcamoid freezed and you were not able to close it and that's the bug you are reporting?

>but if it's true only the window manager controls it then (as long as you test in Windows)

As said before, there isn't any code controlling the window drawing. If there is a bug, it should be maybe in Qt code or totally external to the Webcamoid.

>this can be closed.

I want first, to totally discard a bug in my side, and at the moment I did not found any evidence of that.

hipersayanX, 2023-05-11:
Issue not reproducible, closing.

closed 2023-05-11 by hipersayanX