webcamoid/webcamoid #574: double free or corruption (!prev)

acheong08, 2022-11-15:
Choose one of the templates below that fit better your issue.

# Reporting a problem?

## Summary

double free or corruption (!prev)
[1]    29571 IOT instruction (core dumped)  webcamoid

## Steps to Reproduce

```
$ cmake -S . -B webcamoid-build
$ cmake --build webcamoid-build --parallel 4
$ sudo cmake --install webcamoid-build
$ webcamoid
```

## Your Environment

* Webcamoid information: Webcamoid 9.0.0 (amd64)
OS: Ubuntu 22.04.1 LTS x86_64 
Host: Laptop AB 
Kernel: 5.15.0-52-generic

# Details
First run: ```
[2022-11-15 20:01:42.923, Webcamoid, 0x7f71bc6667c0,  (0)] info: Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[2022-11-15 20:01:43.734, Webcamoid, 0x7f71bc6667c0, VideoOutputPicture.qml (77)] warning: qrc:/Webcamoid/share/qml/VideoOutputPicture.qml:77:13: QML Image: Protocol "" is unknown
[2022-11-15 20:01:43.743, Webcamoid, 0x7f71bc6667c0, main.qml (152)] warning: qrc:/Webcamoid/share/qml/main.qml:152:5: QML ColumnLayout: Binding loop detected for property "width"
[2022-11-15 20:01:43.788, Webcamoid, 0x7f71bc6667c0, main.qml (152)] warning: qrc:/Webcamoid/share/qml/main.qml:152:5: QML ColumnLayout: Binding loop detected for property "width"
[2022-11-15 20:01:43.859, Webcamoid, 0x7f7101ffb640,  (0)] debug: Camera input frame format: AkCaps(mimeType="video/unknown",fourcc=QVariant(QString, "MJPG"),fps=QVariant(QString, "60/1"),height=QVariant(uint, 240),width=QVariant(uint, 424))
free(): invalid next size (normal)
[1]    30612 IOT instruction (core dumped)  webcamoid
```
Second run: ```
[2022-11-15 20:01:47.832, Webcamoid, 0x7f629313e7c0,  (0)] info: Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[2022-11-15 20:01:47.902, Webcamoid, 0x7f629313e7c0,  (0)] info: "An instance of Webcamoid is already running"
```
Third run:
```
[2022-11-15 20:02:37.220, Webcamoid, 0x7fd3e302b7c0,  (0)] info: Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[2022-11-15 20:02:37.947, Webcamoid, 0x7fd3e302b7c0, VideoOutputPicture.qml (77)] warning: qrc:/Webcamoid/share/qml/VideoOutputPicture.qml:77:13: QML Image: Protocol "" is unknown
[2022-11-15 20:02:37.958, Webcamoid, 0x7fd3e302b7c0, main.qml (152)] warning: qrc:/Webcamoid/share/qml/main.qml:152:5: QML ColumnLayout: Binding loop detected for property "width"
[2022-11-15 20:02:38.005, Webcamoid, 0x7fd3e302b7c0, main.qml (152)] warning: qrc:/Webcamoid/share/qml/main.qml:152:5: QML ColumnLayout: Binding loop detected for property "width"
[2022-11-15 20:02:38.145, Webcamoid, 0x7fd310965640,  (0)] debug: Camera input frame format: AkCaps(mimeType="video/unknown",fourcc=QVariant(QString, "MJPG"),fps=QVariant(QString, "60/1"),height=QVariant(uint, 240),width=QVariant(uint, 424))
double free or corruption (!prev)
[1]    31135 IOT instruction (core dumped)  webcamoid
```

acheong08, 2022-11-15:
With appimage:
```
[2022-11-15 20:04:20.004, Webcamoid, 0x7feb227fc640,  (0)] debug: Camera input frame format: AkCaps(mimeType="video/unknown",fourcc=QVariant(QString, "MJPG"),fps=QVariant(QString, "60/1"),height=QVariant(uint, 240),width=QVariant(uint, 424))
malloc(): unaligned tcache chunk detected
Aborted (core dumped)
```

hipersayanX, 2022-11-20:
```
[2022-11-15 20:04:20.004, Webcamoid, 0x7feb227fc640,  (0)] debug: Camera input frame format: AkCaps(mimeType="video/unknown",fourcc=QVariant(QString, "MJPG"),fps=QVariant(QString, "60/1"),height=QVariant(uint, 240),width=QVariant(uint, 424))
malloc(): unaligned tcache chunk detected
Aborted (core dumped)
```

I bet this is probably due to the frame alignment (width == 424), it should be fixed already in the [video-convert branch](https://github.com/webcamoid/webcamoid/tree/video-converter).

closed 2022-11-21 by acheong08