webcamoid/webcamoid #575: Window scaling on small screen (Steam Deck)

Vitnim, 2022-11-24:
Trying to use Webcamoid on a Steam Deck.  SteamOS runs a modified version of Arch Linux, I believe.  The Webcamoid window does not scale properly on the small 1280x800 screen.  I tried adjusting the DPI using xrandr, but that also seems to have no effect.  This is the only application I have run *so far* that does not scale appropriately.

The buttons and accompanying text in the window are huge.  There are also no scroll bars, so it is impossible to interact with the settings beyond the first one or two buttons on each level. 

Where does Webcamoid look for scaling settings and/or is there a simple fix that is beyond my basic understanding?  
Thanks

DanH42, 2022-11-26:
> The buttons and accompanying text in the window are huge.

I'm just a normal user and can't offer any real help, but I believe the button/text size you mentioned may be intentional. Version 9.0 was designed with "[inspiration from phone camera apps to give a similar experience to the user](https://webcamoid.github.io/posts/2022/02/15/webcamoid-900-released/)," so interactive elements look like they've all been scaled to be tapped with a finger on a phone-sized screen, even if you're on a normal computer with a high-resolution screen and a mouse. For reference, here's how it looks on my laptop with a 2256x1504 display running fairly stock KDE Plasma 5.22 with DPI scaling disabled:
[image]
As you can see, I've got working scrollbars, so that part definitely sounds like a bug.

I do think I'm seeing the same DPI scaling issue, though: with my scale factor set at 125%, as I usually have it on this machine, the Webcamoid controls remain the same size, and the default window size is just larger:
[image]

In either of these screenshots, if I imagine an Android system UI wrapped around the window instead of a desktop-style titlebar, everything makes a lot more sense. As-is, it looks pretty out-of-place next to any of my other applications. Depending on how the DPI scaling issue shakes out, maybe a "touch-friendly UI" settings toggle would be a useful addition.

hipersayanX, 2022-11-26:
The UI is based in the [Material style](https://m2.material.io/components/buttons#specs) design, and everything is calculated using dp units using these formulas (which seems to be [pretty common](https://www.altova.com/manual/MobileTogether/mobiletogetherdesigner/mtdobjsfeatures_sizes.html)):

```
dp = 160 * px / dpi
px = dp * dpi / 160
```

And, at least for me, it worked fine in both desktop and mobile, but in HDPI screens it seems to be causing  problems. Actually, it's possible to [scale the controls internally](https://github.com/webcamoid/webcamoid/blob/26995d8c3850af648ecd91b5e3eb11b1a415d12c/libAvKys/Lib/src/qml/aktheme.h#L35), I'll need to add an option to control that from the configs or an environment variable.

Vitnim, 2022-12-03:
Thank you for the insight and attention.  Keep in mind that the Steam Deck is a very modern device with a very low-resolution screen, so it is likely the exception to the rule.  However I can also see it being an issue on low-res screens on Raspberry Pi's, for example.  
DanH42 - The scrollbar on your window appears to be part of the program (scrolling through a menu), whereas there is no scrollbar as part of the window.
I will look for a future update to include an option to scale.  In the meantime, is there any way I might affect the scaling externally?  I don't know enough to understand why changing the system dpi would not affect the dp formulas you mention.

hipersayanX, 2022-12-04:
I've added support for QT_SCALE_FACTOR in [81eba0d](https://github.com/webcamoid/webcamoid/commit/81eba0dca0bfb8244448810744b7725d463bab48), which should allow to scale down the UI controls manually. Also added the rule to make the controls smaller when the window width or height is smaller than 10cm, [here](https://github.com/webcamoid/webcamoid/commit/81eba0dca0bfb8244448810744b7725d463bab48#diff-e2a1ed98d67b4f529d535b417427f7e9ad7df3c9c4518e6145f67894dcafde30).
Since I haven't a Steam Deck to test, I've tried imitating the proportions. This is how Webcamoid looked before the patch.

[image]
[image]

And this is how it looked after it.

[image]
[image]

Based on @DanH42 captures, it should look much better now.
If someone is interested in testing and giving feedback, the changes were applied in the [video-converter branch](https://github.com/webcamoid/webcamoid/tree/video-converter).

rokejulianlockhart, 2023-01-24:
Does the desktop version use different scaling to the Android version? Considering (per what I recall from some time ago) that the Android version appears to not scale too hideously (at least, when compared to that screenshot) perhaps how it scales should be investigated as a potentially easier way to remediate this problem.

hipersayanX, 2023-06-10:
Should be fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX